ftctechnh/ftc_app #157: Core Device Interface not showing up

afilini, 2016-03-01:
Hi,
we are testing out our Core Device Interface from Modern Robotics, but it does not get recognized by the normal "Scan" process (even though the green power LED is powered on). The module shows up in the Core Device Discovery application, so I think this might be Android-related issue. Is there a particular procedure we can use to test it out and fill a more accurate bug report?

Also, what is the XML hardware map tag name for the Core Device Interface? We could also add the entry by hand to our file.

This is the Core Device Discovery screenshot:

[image]

nrizzio, 2016-03-01:
This repository is for the FTC SDK and is maintained by FIRST. FIRST didn't make Core Device Discovery, so I would recommend you get help from Modern Robotics.

vwestin, 2016-03-02:
Once you have the LED showing up on the module, it should be visible to the code in the phone. We had other challenges when the LED on the module was not lighting, but once it is you should be good. We found that it did show up in the scan from the phone at that point if you have the right version of library on the phone (anything at/after library 15.x seems to work).

Note that our attempts to use a name other than the default "Device Interface Module 1" generated errors, so we have left it alone. The XML entry looks like this (your serial number will be different, of course):
    <DeviceInterfaceModule name="Device Interface Module 1" serialNumber="AL00VFOS">
    </DeviceInterfaceModule>

We use the 'adb pull' and 'adb push' commands to the XML files on the phone in /sdcard/FIRST/ to allow us easy edits of the XML data. A pull of a new scan to 'fred.xml' is a simple way to get the correct formatting with each device serial number and all ('adb pull /sdcard/FIRST/fred.xml .').

We have the Core Device Interface module working as a part of our code. There are a few items you will need:

- In the package, you need to import the definitions:
import com.qualcomm.robotcore.hardware.DigitalChannelController;

- In the OpMode class, you need an instance:
DigitalChannelController digitalInterface;

-- For our model, we also chose to define constants for the channels we will use:
public static final int LED_RED_CH = 1;
public static final int LED_GREEN_CH = 2;
public static final int LED_BLUE_CH = 3;

- In the init code, you need to map the instance
digitalInterface = hardwareMap.deviceInterfaceModule.get("Device Interface Module 1");

--- As an example, we are using digital 5V ports to drive relays to control LEDs:
digitalInterface.setDigitalChannelMode(LED_RED_CH, DigitalChannelController.Mode.OUTPUT);
digitalInterface.setDigitalChannelMode(LED_GREEN_CH, DigitalChannelController.Mode.OUTPUT);
digitalInterface.setDigitalChannelMode(LED_BLUE_CH, DigitalChannelController.Mode.OUTPUT);

-- And then in your code, you can interact with those channels:
digitalInterface.setDigitalChannelState(LED_RED_CH, newRed);
digitalInterface.setDigitalChannelState(LED_GREEN_CH, newGreen);
digitalInterface.setDigitalChannelState(LED_BLUE_CH, newBlue);

I hope this is useful.

Vince Westin
Coach, Rockin' Robots #5940

afilini, 2016-03-03:
Hi Vince,
as I said before, we are using the latest version available, I merged all the last commits into our repo. 
Did you actually forget to add the xml tag? I don't see it.. 

[image]

Thank you really much

vwestin, 2016-03-06:
Sorry about that. The XML looks like this (with your SN, obviously):
    <DeviceInterfaceModule name="Device Interface Module 1" serialNumber="AL00VFOS">
    </DeviceInterfaceModule>

vwestin, 2016-03-19:
The XML is getting hidden by the software running this site. It may be trying to interpret it as HTML, and failing that displays nothing. Here is the code with the ">" changed to "]":
    [DeviceInterfaceModule name="Device Interface Module 1" serialNumber="AL00VFOS"]
    [/DeviceInterfaceModule]

Now I know why I thought I had it in my first post but did not see it looking back. In edit mode on either of my posts above, the XML code is present. It just does not display in the final version.

closed 2017-05-06 by afilini