ftctechnh/ftc_app #160: OpMode Init commands are not consistent

vwestin, 2016-03-06:
We have noticed that when we give certain commands in init, for servo positions and for the digital outputs on the device interface module specifically, they do not always take effect. They will take effect when start is pressed, but sometimes not before. As an example, we have LEDs attached to relays on the device interface module. They are programmed to come on with our alliance color during init in each OpMode. However, after a reset of the phone/robot, we find that they often do not.

For example, after a reset we can select the Autonomous OpMode. We run init, but no LEDs are activated. We stop the OpMode and change to the TeleOp OpMode. We run init, and now the LEDs come on. We stop it and go back to the Autonomous OpMode. When we run init again, the LEDs are on (and will change color if the other OpMode had them set differently). The act of changing between OpModes and doing the init causes the activity to act correctly.

Note that we do have 7 devices plugged into the USB power on our PDM (4 motor controllers and all 8 motors, with 6+ running at once, 1 servo controller, 1 legacy module, 1 device interface module). We are having intermittent issues of USB devices not being found, which are very reminiscent of the challenges we had early on with the PDM when we went above 3 devices attached. The RMA of the early devices has had us fairly stable, but we are planning to move some of our controllers back to legacy versions to see if it helps our stability. Our robot with all legacy motor controllers has had ZERO issues with phone/PDM stability.

Please let us know what additional documentation would be useful to assist in debugging this situation.

gearsincorg, 2016-09-04:
Hi.  Can you test to see if this is still an issue with the new 2.X code release?  
It should have been eliminated, but your hardware configuration is .... "extensive".

cmacfarl, 2016-11-14:
Closing per inactivity.  If this is still an issue with the latest beta, please feel free to reopen.

closed 2016-11-14 by cmacfarl