ftctechnh/ftc_app #164: FTC Technology Forum down?

AlecHub, 2016-03-31:
I get the following error when trying to access http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology:

> **Warning:** Illegal string offset 'name' in **[path]/includes/functions.php** on line **6988**
> Unable to add cookies, header already sent.
> File: /homepages/4/d174712211/htdocs/public_html/ftcforum/includes/class_core.php
> Line: 5615

Anyone else getting this error?

magneticflux-, 2016-03-31:
@AlecHub I am as well. Windows 8.1 64x, Chrome 49.0.2623.110

lukethecoder64, 2016-03-31:
Same here. Firefox.

AlecHub, 2016-04-01:
I am able to access the technology forum only if I login to http://ftcforum.usfirst.org.

**Workaround:** login to http://ftcforum.usfirst.org

AlecHub, 2016-04-05:
It is no longer necessary to login to access the forum, so the issue appears to be resolved.

calebsander, 2016-05-31:
Can this be closed?

lukethecoder64, 2016-07-07:
Could this be closed?

closed 2016-07-19 by ftctechnh