ftctechnh/ftc_app #166: HiTechnic servo controllers do not properly enable PWM between runs of an OpMode

calebsander, 2016-04-03:
The issue is with this piece of code in `HiTechnicNxtServoController.setServoPosition()`:
````java
if(var4 != this.a[OFFSET_SERVO_MAP[channel]]) { //if position is new from last time
  this.d = true; //indicate that a change has occurred and positions should be resent
  this.a[OFFSET_SERVO_MAP[channel]] = var4; //request that position be set to new position
  this.a[10] = PWM_ENABLE; //request that PWM be enabled
}
````
If one `OpMode` sets all the servo positions to a certain value and is then quit, the PWM is disabled. When that `OpMode` or another one is then run and the servos are set to the same position as they were at the end of the previous `OpMode`, the `HiTechnicNxtServoController` object assumes no change has occurred and never re-enables the PWM. The `this.a[10] = PWM_ENABLE` should occur regardless of whether the position has changed since the previous time.

rgatkinson, 2016-04-04:
Thanks Caleb. Did you actually run across this in practice, or is this a thought experiment? And which version of the library were you looking at (I don't believe this is a problem in the latest version: pwmDisable() invalidates the cache).

calebsander, 2016-04-04:
We were experiencing this issue in practice and I traced it to this problem. I hackily resolved it by having each call to set the position in our `ServoWrapper` class set a different position (`(pos + 0.5) % 1.0`) first. This was in 1.6, so it could very well be resolved in 1.7.

rgatkinson, 2016-04-04:
Thanks. Yes, this was broken in 1.6.

calebsander, 2016-04-04:
A brief look over the 1.7 beta code suggests that this issue is fixed by the invalidation feature. We downgraded from 1.7 because we were getting some unexplained connection issues and robot controller shutdowns, so I can't really test at the moment.

batteriesinblack, 2016-04-08:
Even we are experiencing unexplained connection issues and robot controller issues. It specifically seems to happen when we try to initialize one of our servo controllers

After testing with multiple servo controllers we have reverted back to SDK 1.6

gearsincorg, 2016-09-04:
Closed, as this issue relates to previous season code.

closed 2016-09-04 by gearsincorg