ftctechnh/ftc_app #178: Android Studio 2.0 Support

2000andrewbowen, 2016-07-28:
I downloaded the sdk on 7/26/2016 and tried to run it, but I got this error: 

Error:Execution failed for task ':FtcRobotController:compileDebugJavaWithJavac'.
> java.io.FileNotFoundException: C:\Users\*\*\*\*\ftc_app-master\FtcRobotController\libs\android-support-v4.jar (The system cannot find the file specified)

I was able to fix it, but it took me an hour to found how to fix the error at this forum [post](http://ftcforum.usfirst.org/showthread.php?6738-Android-Studio-2-0-and-SDK-Release-16-03-09-Signed-Gradle) . 

Please change fix this.

JacobAMason, 2016-07-28:
This is already being addressed at #176

cmacfarl, 2016-08-20:
Fixed in SDK 2.0 (in the beta branch currently)

closed 2016-08-20 by cmacfarl