ftctechnh/ftc_app #180: Driver Station UI issues with autonomous programs

calebsander, 2016-08-24:
1. When stopping an autonomous program, the center button returns to displaying a stop button.
2. When stopping an autonomous program being timed, the time keeps counting down until tapping on the `OpMode` list. When initializing the autonomous program again, it continues counting down from where it left off.
3. When the timer does expire, the program stops but the center button returns to a stop button and the remaining time starts counting down from 30 again and is displayed over the stopwatch icon.

rgatkinson, 2016-08-24:
Thanks for the report, Caleb. I've tried to reproduce these issues, but I've been unable to (on my ZTE Speeds). Might it be possible to get further details as to the steps you went through that exhibited the problems?

calebsander, 2016-08-24:
I created a video showing the steps I took: https://youtu.be/WRfRVmKgAs8. I also found that the OpMode list is empty when first loading the apps. I wasn't able to reproduce the 3rd issue, but I reproduced the other 2 just fine. I am not sure whether the first issue also occurs with TeleOp programs

rgatkinson, 2016-08-25:
Thanks again. The blank opmode list I understand, and we have a fix for (the interim workaround is to start the DS first, or do restart robot as you showed), but the other two still puzzle me (though I'm away from my phones and can't test thoroughly). Could you possibly share the opmode that you were demonstrating here? Also (if you have the time): does it repro for you using the sample telemetry opmode?

gearsincorg, 2016-08-25:
Don't know if it's related but.... I tripped myself up on auto today.
I had code that was waiting for the opmode to start:

        while (!isStarted()) {
            telemetry.addData(">", "Robot Heading = %d", gyro.getIntegratedZValue());
            telemetry.update();
        }

Because I didn't have an idle(); call in the loop, it would do weird things if I tried to do something other than "play" once I had done an init.  (eg: make another selection while init was active).

calebsander, 2016-08-25:
Here is a gist that should have all the classes I was using in my test `OpMode`: https://gist.github.com/calebsander/32793b920f832d271bd6a8b1c36162d4. It is not a `LinearOpMode`, but I suspect that the issue might be caused by the relatively expensive loop operations (adding up the number of blue in every pixel of a 120x160 image).

rgatkinson, 2016-09-20:
@calebsander: is this still an issue? I think it might be fixed in 2.2.

calebsander, 2016-11-05:
Seems fixed in newer releases

closed 2016-11-05 by calebsander