ftctechnh/ftc_app #187: Request: Access to camera when using Vuforia

khoyt5, 2016-09-07:
Last year our team experimented with using the camera and taking a picture of the beacon and then process the pixels to determine which button to push based on Red color or Blue color.

The Vuforia capability that is now the SDK looks promising for use but I could see using the raw pixel data in parallel. Is it possible to update the SDK calls to Vuforia to take a current snapshot and provide the pixel data to the running OpMode?

rgatkinson, 2016-09-09:
This is a very reasonable and understandable request. Unfortunately, it's not technically possible to achieve, at least with the currently available version of Vuforia.

rgatkinson, 2016-09-14:
Update: it turns out that this may be possible after all (thanks FTC 3491!). Investigation is proceeding...

rgatkinson, 2016-09-15:
There are two solutions to this. First, a hack that will work with the existing release (but may not be supported long term), and second, an architected solution that should go out in the next release.

Both approaches ultimately provide access to a series of underlying Vuforia [Frame](https://library.vuforia.com/sites/default/api/java/classcom_1_1vuforia_1_1Frame.html) [objects](https://library.vuforia.com/sites/default/api/cpp/classVuforia_1_1Frame.html) from which camera image data can be accessed (though, in my experience, only monochrome data, not color).

The first approach is to implement the following class:

```
    class VuforiaLocalizerImplSubclass extends VuforiaLocalizerImpl
        {
        class VuforiaCallbackSubclass extends VuforiaCallback
            {
            @Override public synchronized void Vuforia_onUpdate(State state)
                {
                super.Vuforia_onUpdate(state);
                Frame frame = state.getFrame().clone(); // clone so frame could live beyond the callback
                RobotLog.vv(TAG, "received Vuforia frame#=%d", frame.getIndex());
                }
            }

        VuforiaLocalizerImplSubclass(VuforiaLocalizer.Parameters parameters)
            {
            super(parameters);
            this.vuforiaCallback = new VuforiaCallbackSubclass();
            stopAR();
            startAR();
            }
        }
```
Modify the callback to do whatever it is you wish to do instead of just writing to the log. Don't do lengthy operations in the callback. Instantiate this class (with ``new``) rather than calling ``ClassFactory.createVuforiaLocalizer()``.

The second solution involves the following addition to the VuforiaLocalizer interface:

```
    /**
     * (Advanced) Returns a queue into which, if requested, Vuforia {@link Frame}s are placed
     * as they become available. This provides a means by which camera image data can be accessed
     * even while the Vuforia engine is running.
     *
     * <p>While the Vuforia engine is running, it takes exclusive ownership of the camera it is
     * using. This impedes the ability to also use that camera image data in alternative visual
     * processing algorithms. However, periodically (at a rate of tens of Hz), the Vuforia engine
     * makes available {@link Frame}s containing snapshots from the camera, and through which
     * camera image data can be retrieved. These can optionally be retrieved through the frame
     * queue.</p>
     *
     * <p>To access these {@link Frame}s, call {@link #setFrameQueueCapacity(int)} to enable the
     * frame queue. Once enabled, the frame queue can be accessed using {@link #getFrameQueue()}
     * and the methods thereon used to access {@link Frame}s as they become available.</p>
     *
     * <p>When {@link #setFrameQueueCapacity(int)} is called, any frame queue returned previously by
     * {@link #getFrameQueue()} becomes invalid and must be re-fetched.</p>
     *
     * @return a queue through which Vuforia {@link Frame}s may be retrieved.
     *
     * @see #setFrameQueueCapacity(int)
     * @see <a href="https://library.vuforia.com/sites/default/api/java/classcom_1_1vuforia_1_1Frame.html">Frame (Java)</a>
     * @see <a href="https://library.vuforia.com/sites/default/api/cpp/classVuforia_1_1Frame.html">Frame (C++)</a>
     * @see <a href="https://developer.android.com/reference/java/util/concurrent/BlockingQueue.html">BlockingQueue</a>
     */
    BlockingQueue<Frame> getFrameQueue();

    /**
     * Sets the maximum number of {@link Frame}s that will simultaneously be stored in the
     * frame queue. If the queue is full and new {@link Frame}s become available, older frames
     * will be discarded. The frame queue initially has a capacity of zero.
     *
     * <p>Note that calling this method invalidates any frame queue retreived previously
     * through {@link #getFrameQueue()}.</p>
     *
     * @param capacity the maximum number of items that may be stored in the frame queue
     * @see #getFrameQueue()
     * @see #getFrameQueueCapacity()
     */
    void setFrameQueueCapacity(int capacity);

    /**
     * Returns the current capacity of the frame queue.
     * @return the current capacity of the frame queue.
     * @see #setFrameQueueCapacity(int)
     * @see #getFrameQueue()
     */
    int getFrameQueueCapacity();
```

rgatkinson, 2016-09-15:
Update: calling
```
Vuforia.setFrameFormat(PIXEL_FORMAT.RGB888, true);
```
will enable delivery of 24-bit color images accessible from the Frame. Analogous calls would also provide images in other formats (see the constants in [PIXEL_FORMAT](https://library.vuforia.com/sites/default/api/cpp/namespaceVuforia.html#a8f7511b96bcb33bc2ea176a1a8dafb59)). You can place this call after the call to startAR() illustrated above.

gebailey, 2016-10-08:
I see in the release notes for v2.30 it says: "Fix deadlock / make camera data available while Vuforia is running."  Does that fix basically allow us to use the second solution as listed above?  I'm trying to square this with suggestions given by FIXIT3491 in their YouTube video explaining how they're able to capture the frames from Vuforia.

rgatkinson, 2016-10-08:
The above described architected solution is now implemented in VuforiaLocalizer.

cmacfarl, 2016-11-14:
Per @rgatkinson's comment, closing.

closed 2016-11-14 by cmacfarl