ftctechnh/ftc_app #195: Enable GitHub Pages so that documentation can be viewed online

nrizzio, 2016-09-17:
All that is required is for @ftctechnh to [enable GitHub Pages in repository settings based off of the master branch.](https://help.github.com/articles/configuring-a-publishing-source-for-github-pages/) After that is done, you can view all FTC Javadocs online without having to clone the repository.
[image]
Here's my fork, for example: https://nrizzio.github.io/ftc_app/doc/javadoc/index.html
After it is enabled, a link to https://ftctechnh.github.io/ftc_app/doc/javadoc/index.html could be added to the readme.

richiksc, 2016-11-23:
Yes please! Host the Javadoc there and take the javadoc out of the repo, then add a download option for the javadoc on the wiki pages. Migrate the PDF tutorials to the wikis. I could do this for you guys if need be.

cmacfarl, 2016-12-07:
Done.  See the readme.  Note that the javadoc was not taken out of the repo, but easy online access is now provided.

closed 2016-12-07 by cmacfarl