ftctechnh/ftc_app #209: Controller and Driver Setup Issues

stemachievement, 2016-09-23:
Unable to setup the Android phones. Error messages..."This file is too big to display" for the following:  doc/apk/FtcRobotController-release.app
Same with the driver file

gearsincorg, 2016-09-23:
Specifically: what are you doing to have this message come up?

File transfer? install? run?
More details please.

stemachievement, 2016-09-23:
Installing the Controller and Driver files on the Android phone for the first time.

magneticflux-, 2016-09-23:
In what application is this? Command line?

stemachievement, 2016-09-23:
- https://github.com/ftcechnh/ftc_app

- jump to file

- doc/apk

- doc/apk/FtcRobotController-release.app

Error Message:
"This file is too big to display"

This is the message that should be displayed:
"This file is binary and cannot be displayed inline. Open binary file"

magneticflux-, 2016-09-23:
You have to download the repo onto your computer as a .zip, and then use the ADB application to install an .apk on your phone.

stemachievement, 2016-09-23:
Ok...thanks I'll try that.

magneticflux-, 2016-09-23:
Do you have Android Studio installed on your computer?