ftctechnh/ftc_app #214: Why does @tomeng70 or @ftctechnh never reply to (or fix) issues?

richiksc, 2016-09-25:
This feels bad for an open source project. The maintainer should always listen to contributors. By submitting an issue, I am contributing to the project (in a way). I could submit a pull request, but that would never get merged/would not follow the style of Tom Eng's code.

This feels extremely unlike Gracious Professionalism. The 69 closed issues were all closed by the submitter or other users, the suggested changes never making it to the actual SDK, only hacky ways suggested by others.

magneticflux-, 2016-09-25:
While you are correct in that most of the issues were closed by the opener, it was primarily because a fix or feature was implemented in either the beta or the full release. See #151 #137 #118 for examples. In addition, much of the code is provided by Qualcomm, which is not controlled by FIRST.

gearsincorg, 2016-09-25:
@richiksc  Your gross simplification of the past history of this project also feels pretty unGracious.

Several issues:
1) This repo is not the development source.  It's a public repo. that is derived from the source as a method to distribute the SDK, and collect feedback.  So contributions will NEVER be incorporated directly into this repo.  They will be incorporated into the development master, and then released with many other changes at a later time.  So don't expect a 1:1 correlation.

2) Most (almost all) of the pull requests have been issued by mistake by people experimenting with git, and learning how to manage their own copies of the SDK. (which is why they have all been deleted by their originators).

3) Any and all legitimate bugs/problems/suggestions are listened to by the developers.  A huge number of these, and other issues have been incorporated from last season's contributions.  

4) I would challenge you to find even a small number of reasonable changes that were not considered and incorporated into this season's SDK.  Enhanced user interfaces, more open sensor interfaces, greater overall reliability, vision processing,  improved robot configuration, integrated inspection processes, more phone choices, more OS options etc.  Even internal code visibility has been improved withing the limits permitted.

AlecHub, 2016-10-07:
@richiksc is correct.  Submitting an issue or feature request is a contribution to the project.

Tickets need to be acknowledged by the developers, yay or nay, in a timely manner; not next season, and not months and months after the fact as has been the practice.    

When the developers provide real-time status of issues or feature requests, teams can be in a position to decide if they should implement workarounds or wait for the release in which the ticket is scheduled to be addressed.

rgatkinson, 2016-10-07:
Do please remember in your requests that virtually all participants here are volunteers, many of which have contributed literally hundreds and hundreds of hours this past year working to improve the SDK.

You can rest assured that all issues and submissions are seen and observed in virtually real time: silence is not an indication of inattentiveness. While many can be addressed quickly, there are some that do require (and, indeed, have required) weeks or months of investigation and contemplation before any meaningful comment can be made thereon.

cmacfarl, 2016-11-14:
Others have addressed the concerns of the original submitter.  Closing.

closed 2016-11-14 by cmacfarl