ftctechnh/ftc_app #215: Unexpected IMU Initialization Error

branvu, 2016-09-25:
When we try to initialize the IMU, an error occurs in the log `E/IMU: unexpected chip: expected=-96 found=0`. 
We are able to see the the Core Device Interface on the Core Device Discovery, and we checked the chipId register and confirmed that it reads the right value. Has anybody else ran into this issue before? We did not have this issue with @rgatkinson 's Swerve library before. Thanks!

rgatkinson, 2016-09-25:
Might it be possible to get a logcat trace that illustrates the problem?

branvu, 2016-09-25:
```
09-25 17:04:14.762 6021-5464/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: 'LinearOpMode main' starting...
09-25 17:04:14.772 6021-6186/com.qualcomm.ftcrobotcontroller V/Robocol: sending CMD_INIT_OP_MODE_RESP(28578), attempt: 0
09-25 17:04:14.812 6021-2317/com.qualcomm.ftcrobotcontroller V/Robocol: received command: CMD_INIT_OP_MODE_RESP(28578) Sensor: DbzAdafruit IMU
09-25 17:04:14.992 6021-2468/com.qualcomm.ftcrobotcontroller D/dalvikvm: GC_FOR_ALLOC freed 2048K, 49% free 6525K/12688K, paused 24ms, total 24ms
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller E/IMU: unexpected chip: expected=-96 found=0
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller W/IMU: retrying IMU initialization: initializeOnce() failed
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller E/IMU: unexpected chip: expected=-96 found=0
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller W/IMU: retrying IMU initialization: initializeOnce() failed
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller E/IMU: unexpected chip: expected=-96 found=0
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller W/IMU: retrying IMU initialization: initializeOnce() failed
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller E/IMU: unexpected chip: expected=-96 found=0
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller W/IMU: retrying IMU initialization: initializeOnce() failed
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller E/IMU: IMU initialization failed
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller D/RobotCore: LinearOpMode received an InterruptedException; shutting down this linear op mode
09-25 17:04:15.312 6021-5464/com.qualcomm.ftcrobotcontroller V/RobotCore: thread: ...terminating 'LinearOpMode main'
```

rgatkinson, 2016-10-07:
I have a good hunch that the (imminent!) next release ought to address this problem. If that turns out not to be the case, please advise and we'll investigate further.

WEEEEE1234, 2017-02-10:
Well, you guys have probably fixed your problem by now, however for the benefit of future teams viewing this issue I would like to add that our team has encountered a similar issue (with the error message saying  `E/IMU: unexpected chip: expected=-96 found=-1` instead, as opposed to `found=0`). We have found that this is usually caused by one of the sensor wires on the Device Interface Module being plugged in backwards. Upon properly reinserting the wire, the error resolved itself.

ldricci3, 2017-02-21:
We are having the same problem, and all of our wires are plugged in correctly... 

Are there any other possible solutions?

cmacfarl, 2017-02-21:
@ldricci3 which version of the SDK are you using?

ldricci3, 2017-02-22:
It turns out we had probably had plugged in the IMU backwards at some point, and thereby fried it for good.

To fix the problem we had to change to a new Adafruit IMU.

The broken IMO was also messing up all the other sensors when plugged in to i2c: they all returned values of 0 or 255 when it was connected. When we switched to the new IMU all the other sensors started working.

We are still having problems starting up autonomous (when we run our autonomous and stop it, we have to physically restart the robot and the app), but the IMU appears to be working now.
We think the problem is related to the DIM, because it works when we don't plug the DIM in, but it is probably a code issue somewhere.

closed 2017-08-31 by branvu