ftctechnh/ftc_app #224: Robot Controller app crashes when trying to load a resource config file that no longer exists

calebsander, 2016-10-05:
This has become extremely annoying in our workflow, in which we often download from a branch with an additional resource config file to a phone, and then later want to download from `master`. Steps to reproduce:

1. Download the app with an included resource config file
2. Select the config file on the phone
3. Quit the app
4. Remove the resource config file from the Android Studio project
5. Redownload the app
6. Watch the app crash on launch with the following error:

````
09-01 09:58:18.403 26835-26835/com.qualcomm.ftcrobotcontroller E/AndroidRuntime: FATAL EXCEPTION: main
   Process: com.qualcomm.ftcrobotcontroller, PID: 26835
   android.content.res.Resources$NotFoundException: Resource ID #0x7f04000b
       at android.content.res.Resources.getValue(Resources.java:1351)
       at android.content.res.Resources.loadXmlResourceParser(Resources.java:2777)
       at android.content.res.Resources.getXml(Resources.java:1214)
       at com.qualcomm.ftccommon.configuration.RobotConfigFile.getXmlResource(RobotConfigFile.java:190)
       at com.qualcomm.ftccommon.configuration.RobotConfigFile.getXml(RobotConfigFile.java:159)
       at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.requestRobotSetup(FtcRobotControllerActivity.java:492)
       at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.updateUIAndRequestRobotSetup(FtcRobotControllerActivity.java:482)
       at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onServiceBind(FtcRobotControllerActivity.java:475)
       at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity$1.onServiceConnected(FtcRobotControllerActivity.java:149)
       at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1223)
       at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1240)
       at android.os.Handler.handleCallback(Handler.java:746)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.app.ActivityThread.main(ActivityThread.java:5443)
       at java.lang.reflect.Method.invoke(Native Method)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)
````

rgatkinson, 2016-10-05:
Thank you for the clear and cogent bug report.

cmacfarl, 2016-10-05:
The workaround is to select a local config prior to pushing the RC built
from a stock SDK.

And yeah, that's not right, and will be fixed.  As Robert said, your bug
report is very helpful  Thank you for that.

cmacfarl, 2016-11-14:
Fixed in the new beta.

closed 2016-11-14 by cmacfarl