ftctechnh/ftc_app #228: Git history has been completely changed

nrizzio, 2016-10-08:
e5b437727b5d969f5cc89a42f8d09c2023464284's commit message is very scary:

> removed the PDF from the doc/tutorial
...
In an effort to save space and reduce download time, the older (and outdated) tutorial documents (FTCTraining_Manual.pdf, FTC_FieldCoordinateSystemDefinition.pdf, FTC_NextGenGuide.pdf, and FTC_ZTE_ChannelChange.pdf) have been removed.

Looking at my [public fork](https://github.com/nrizzio/ftc_app), it appears that the PDFs have been removed from older commits and every commit thereafter rebased, which has caused a discrepancy in the git commit IDs for the 33 most recent commits in my repository.

This is a big problem for teams like mine, because we incorporate changes by running a `git merge` from our local master to @ftctechnh's remote master. Since the histories have now diverged, nothing merges cleanly, and our repositories will not actually become smaller unless we destroy and re-create our repository.

@ftctechnh should have provided more warning that this change was going to occur, or at least created a separate repository with the lessened file size for the time being while keeping this one as it was, to give teams ample time to transition.

Relates to #71
See https://git-scm.com/book/en/v2/Git-Branching-Rebasing#The-Perils-of-Rebasing
> The Perils of Rebasing
Ahh, but the bliss of rebasing isn’t without its drawbacks, which can be summed up in a single line:
**Do not rebase commits that exist outside your repository.**
If you follow that guideline, you’ll be fine. If you don’t, people will hate you, and you’ll be scorned by friends and family.
When you rebase stuff, you’re abandoning existing commits and creating new ones that are similar but different. If you push commits somewhere and others pull them down and base work on them, and then you rewrite those commits with git rebase and push them up again, your collaborators will have to re-merge their work and things will get messy when you try to pull their work back into yours.

EtherTyper, 2016-10-08:
I completely agree. Performing a `git merge upstream/master -s ours` to keep up to date results in 33 duplicate commits in my repository, https://github.com/ethertyper/ftc_app.

EtherTyper, 2016-10-08:
Also related to their rebase issues, why has commit e5b437727b5d969f5cc89a42f8d09c2023464284 not been pushed to the beta branch yet?

clwillingham, 2016-10-10:
After this season is over it may be a good idea to make a brand new git repository on GitHub and move the code over to that repository (without PDF's) and deprecate this repository with a link to the new one in the root directory Readme and in the description to move everyone over to that repository. 
This is a solution I have seen used in other projects with large repositories before, but it would be a bad idea to make any more large changes to this repository before the season is over, as teams are depending on this repository right now.

There might not be a any good solution to the current issue except possibly finding an old copy of the repo with the history that's been removed, re-adding the PDF's then removing them correctly, but either way, the git repository will not shrink. It's a bit too late to decrease the size of the current repository though, so that may be the best solution, and that solution unfortunately would mess up anyone who's already resolved the merge issues within they're own fork of the repository. And i'm also not sure if that would result in more duplicate commits

EtherTyper, 2016-10-10:
@clwillingham Anyone who's already merged the two histories (like myself) will not experience more duplicate commits or merge conflicts, regardless of which history @ftctechnh chooses to support. Also, external copies of the repository won't be necessary to revert the branches to their prior state, because they forgot to update tags to the No-PDF history, so the old commits still exist.

ftctechnh, 2016-10-10:
Hey All - apologies for breaking the history of the repo while pruning the PDF tutorials out of the repo.  I did not realize that filter branch process would rebase the entire repo and abandon the existing (remaining) commits.

jphigham, 2016-10-11:
filter-branch is definitely a powerful command, and is probably why the section in the [Pro Git book](https://git-scm.com/book/en/v2) is entitled [The Nuclear Option: filter-branch](https://git-scm.com/book/en/v2/Git-Tools-Rewriting-History#The-Nuclear-Option:-filter-branch)! Let us know whether the plan is to proceed with the new history - in meantime I'll hold off updating our team repo.

hexafraction, 2016-10-12:
Why is storage of large files not delegated to [git LFS](https://git-lfs.github.com/)? The fact that updating involves fetching a bunch of intermediate commits containing many binary blob versions, to be locally resolved, is somewhat difficult, especially for teams who either have slow Internet access at their venue, or deal with metered access. AFAIK if LFS is used then a bunch of commits updating file pointers would need to be downloaded and resolved, and then only the newest file revision downloaded from the server.

rgatkinson, 2016-10-12:
You might know more about LFS than I. Suppose I install git and git lfs (they are separate installs; the latter isn't built into the former, I undrestand) and I make a repo with some of its (say) .PDF content in lfs. Some else then clones my repo, but they only have git, not git+gitlfs installed. What do they see? Do they see the .PDF content? Or do they see only the 'pointer files' to the PDF content?

EtherTyper, 2016-10-12:
@hexafraction They tried to migrate PDFs to GitHub Releases, instead of LFS, and ended up duplicating 33 commits, which is what this issue concerns. I agree that LFS would be a good solution if they need to do something like this in the future, so you only had to download the 70MB of PDFs when you (optionally) run `git lfs install`.

EtherTyper, 2016-10-12:
@rgatkinson They get everything that does not show up in `git lfs ls-files` (the LFS-tracked files).

EtherTyper, 2016-10-13:
@rgatkinson You can see what an LFS hosted repository would like on the `lfs` branch of my repository https://github.com/ethertyper/ftc_app. It is a prime example of what would happen if you cloned the repository without access to LFS or the LFS-hosted assets, because GitHub refuses to let public forks (like mine) host LFS assets, as @technoweenie comments on github/git-lfs#1449.

hexafraction, 2016-10-13:
> because GitHub refuses to let public forks (like mine) host LFS assets, as @technoweenie comments on github/git-lfs#1449.

That shouldn't be an issue for actual use though, since once ftctechng/ftc_app has LFS then the forks should also have functioning LFS.

EtherTyper, 2016-10-13:
@hexafraction That's right. It shouldn't be a production problem once @ftctechnh enables it on their repository.

nrizzio, 2016-10-13:
Question: can we retroactively mark certain files as being LFS attachments without changing any commit-ids?

EtherTyper, 2016-10-13:
@nrizzio Yes, but that would not reduce repo bloat. We could suggest they use that .ethos for future files, however.

nrizzio, 2016-10-13:
I'm most concerned about getting my repositories' commit ids back on track. Unless @ftctechnh restores the repository to its form before e5b437727b5d969f5cc89a42f8d09c2023464284, I'll have to change my remote to point towards my unupdated public fork, which is not what I want to do. Could we please get a repository somewhere that keeps established commit history while also being updated with new changes as they are released?

EtherTyper, 2016-10-13:
@nrizzio The line of bash I embedded in my first comment here should get your team's repo on top of both branches (before and after rebase) if you have all changes from the first one, which is still being stored at 06632cb752f0e1e4b673a5581ce6df62b5d8519d on my repo, https://GitHub.com/EtherTyper/ftc_app

hexafraction, 2016-10-16:
Cloning with a depth of 1 is a stopgap and still requires that teams trying to merge changes download all revisions since the last upstream revision they have. There's no way for git to "squash" those commits on the server side in a manner conducive to merging/rebasing.

EtherTyper, 2016-10-16:
@hexafraction That's the point. New teams only have to download the latest commit, and old teams have to download all newer revisions than they have, just like before.

magneticflux-, 2016-10-16:
What's the problem with merging the history in? It added a few commits to my repository, but so what? It's a one-time thing, and now I have the newest history. All it took was 10 minutes of merging using the merge tool built into Android Studio, and added a few extraneous commits that were quickly washed away by team development. It's as simple as running `git merge upstream/master` and accepting the updated binaries and documentation. I would be appreciative if someone could explain all their reasoning for  staying with the old history in one place.

EtherTyper, 2016-10-16:
@magneticflux- I did merge the upstream rebased history, by running the simple git merge strategy `git merge upstream/master -s ours`. However, I was initially confused by the  upload of _**35 commits**_ without changes. It would be easy for someone to figure out what happened and how to fix it by reading this thread, but First gave no indication themselves.

Also, it would be challenging to merge in the same way we did if someone merged their history later. Because git wouldn't be able to find corresponding commits between the two branches, it would produce merge conflicts between the 33 duplicate commits, and new ones, so merging with the ours strategy would actually lose new changes in the repository.

magneticflux-, 2016-10-16:
@EtherTyper I didn't use `-s ours` since there were updates, and I got ~35 duplicate commits. My merge created a commit that resolved my changes with the base repository, and nothing was lost. The Android Studio GUI for merging is useful because it shows lists of files needing to be merged so I can quickly accept changes for stuff I didn't change, and merge stuff that I did change.

EtherTyper, 2016-10-16:
@magneticflux- Cool! I didn't know IntelliJ had merge-conflict resolution functionality built right in... In that case, the only problem with merging the history in is telling people how to.

AlecHub, 2016-10-17:
> the information in the pdfs should be moved to GitHub's Wiki.

I agree with @JacobAMason.  A PDF is an antiquated method of publishing documents.  A Wiki is far superior, especially for software engineering type documents.  For one thing a Wiki has a history so you can easily tell what changed from one revision to another.  Also, a Wiki document takes up far less bits than a pdf document, and will merge effortlessly from one branch to another. 

I volunteer to convert the PDF files to MD files.

closed 2016-10-19 by nrizzio

Schemetrical, 2017-02-25:
So the repo is still bloated and now more so at 400MBs, can we do anything about it?