ftctechnh/ftc_app #230: Provide reliable access to every I2c datum read

rgatkinson, 2016-10-12:
The intent here would be to address the following:

"Quote Originally Posted by hexafraction:
An async implementation should have some kind of process/accumulate callback that is guaranteed to be called for every read made to physical hardware. There are some devices that zero some registers tracking some value whenever read; if these values are ever dropped without being processed the sensor value would be incorrect."

A very easy approach is to similar to what was just added to capture Vuforia Frames. In the I2c case, that would manifest itself as a BlockingQueue of either fixed or infinite capacity into which a record of every read ever made would be placed. The record would have the data itself, the read timestamp, the register index and count, and the i2c address read. Being a BlockingQueue, clients can either manually poll for new data records or set up a thread which would be awakened when new data arrived.

hexafraction, 2016-10-12:
Is there any necessity for this to be a separate thread that needs to be awakened? If the callback is sufficiently light it could be called within an existing I2C-related thread. Of course, that callback would need to be thread-safe (e.g. for a simple summing accumulator, using `AtomicInteger#addAndGet`).

magneticflux-, 2016-10-12:
Both solutions could be put into practice, and it would be possible to write an adapter class to make one appear identical to the other. I think having a BlockingQueue would be the best since it's a lower-level option, and an adapter class should be provided to accept Runnable or Runnable-like objects if the code to run is concise enough.

rgatkinson, 2016-10-12:
Thanks for the feedback!

I hear you about the awaking of a thread, but I am quite reticent to build explicit threading implications into the SDK API model, as would be necessitated by a direct callback approach (on what thread is the callback made? what are the deadlock implications for taking such-and-such a lock on that thread? what are the performance implications to the rest of the system of the callback taking an unreasonable amount of time?). That on top of the fact that concurrent programming _is_ rocket science, and just having to say the callback has to be thread-safe would make it 10x harder to use than it otherwise would be. 

The BlockingQueue approach seems to nicely sidestep these issues: it allows a nice thread-based dedicated producer-consumer relationship for those that understand threading while also supporting a manual polling approach for those who prefer a simpler, single-threaded model.

hexafraction, 2016-10-13:
I feel like these approaches can still be unified. The act of either storing the value to a single variable (that is, not guaranteeing every datum is seen) or queueing it up is itself something that can be implemented as a callback to the lower level I2C code. The exact nature of the call back depends on whether all data is needed, whether it can be quickly processed, and whether it needs to be timestamped and ordered.

magneticflux-, 2016-10-13:
@hexafraction , I think @rgatkinson is worried about an unsuspecting user not reading documentation throughly enough to understand which method to use, and causing the low-level code to hang in either an accidental infinite loop or become ill-behaved. It's understandable, since I'm sure there are many people who didn't study the code before they used it, thinking it would be "obvious" how it should be used. This just forces the user to take responsibility for breaking something instead of possibly breaking some low-level hidden code not meant to be seen under normal circumstances.

hexafraction, 2016-10-13:
> thinking it would be "obvious" how it should be used. This just forces the user to take responsibility for breaking something instead of possibly breaking some low-level hidden code not meant to be seen under normal circumstances.

That brings up a good point. I didn't consider it at first since my mindset is that the "obvious" way out is that I can't assume thread-safety in a callback I write unless I'm told otherwise, and the name/signature of a callback provides an insight into how it should be structured.