ftctechnh/ftc_app #234: Robot Controller (Phone) only detects one servo/motor controller at a time

MEDavy, 2016-10-14:
This has been happening to my team ([8971 Diamond Blades](https://github.com/FTC8971/ftc_app_2016-17)) ever since we started this season.  When we try to add a configuration file, only one of our servo or motor controllers shows up in the list at a time. The problem is almost certainly not with the hardware because (a) this happens with every one of the two Power Distribution Modules, 2 Servo and 5 Motor Controllers, and dozens of USB cables we have tried, and (b) we still have two phones running an older version of the code (from last season) and they do not have this error when using the same hardware. It is also not a problem with our code, like we thought at first, because the issue also occurs with the code straight from the ftctechnh/ftc_app repository. The Controller that does appear in the list is the one that was attached most recently (e.g. we have a servo controller and a motor controllers plugged in and only the servo controller is detected. We unplug and reattach the motor controller and only _it_ shows up in the list _not_ the other.)

KingofGamesYami, 2016-10-14:
This is a known issue with Android 5.x, which is why it is not a legal OS.  You must upgrade to Android 6.x

MEDavy, 2016-10-14:
> This is a known issue with Android 5.x, which is why it is not a legal OS. You must upgrade to Android 6.x

Thanks. I do not actually know what version the phones are running... can't believe I didn't think to check. Don't have access to the phones at the moment but will check at our next meeting.

MEDavy, 2016-10-17:
The Android version was in fact 5.0.2

closed 2016-10-17 by MEDavy