ftctechnh/ftc_app #235: Team Code Not Available Error

NutJobs, 2016-10-15:
We are getting an issue when putting code onto the phone.  A red box appears in the lower left corner saying" error running team code: default activity not found".  We have been working on trying to fix this, but no matter what we do it comes up.  Any ideas?

cmacfarl, 2016-10-15:
Can you take a screenshot of Android Studio when you attempt to push the
app to the phone?  And then describe in detail the steps you take to do so?

NutJobs, 2016-10-16:
Here is what we are doing when we get the error.

We plug in the phone with the Robot Controller installed on it and wait for it to connect. After it connects we press the green triangle run button or the debug run button beside it, and android studio instantly throws the  " error running team code: default activity not found" error box. 
[image]

magneticflux-, 2016-10-16:
@NutJobs You seems to have the ftc_app-master directory inside your project. Try deleting that.

KingofGamesYami, 2016-10-16:
@magneticflux- that's normal for the `Project Files` tab.  Now, if they were on the `Android` tab I would be worried.

cmacfarl, 2016-10-16:
Your build is failing.  The entire build log might shed some light.

In the image you can only see the last bit.  Although it sounds like your
SDK is corrupt.  Have you tried downloading a fresh zip file?

Craig

magneticflux-, 2016-10-16:
@KingofGamesYami Ahh, I see. I'm typically in "Project" or "Android" so I was pretty confused about that extra folder.

NutJobs, 2016-10-17:
We just downloaded a fresh zip.  This was a problem before and after.  Very frustrating.

KevinEmery, 2016-10-17:
@NutJobs Is this problem persisting without you adding any of your code? By that, I mean follow the following steps.

 * Download a fresh zip file
 * Import that as a new project into Android Studio
 * Try to deploy the new project to your phones

If that fails, then it means this is a problem with either the project or your environment. If that succeeds (and then fails when you add your team's code to the project) then it is more likely an issue with your code.

cmacfarl, 2016-10-17:
Can you please grab the entire build log, post it here and describe all of
the steps you are taking to download, open, and build your project?

cmacfarl, 2016-11-14:
Please reopen if you are still seeing the problem.  Otherwise, closed.

closed 2016-11-14 by cmacfarl