ftctechnh/ftc_app #249: The APK is not generating.

LemonJustice, 2016-10-28:
I'm trying to build an APK from a sample project and instead of generating a APK, it is generating an AAR.  Is there a way to fix this?

cmacfarl, 2016-10-28:
You are going to need to be much more descriptive than this if anyone is to help you.

#### How Do I Submit A (Good) Bug Report?

Explain the problem and include additional details to help maintainers reproduce the problem:

- Use a clear and descriptive title for the issue to identify the problem.

- Describe the exact steps which reproduce the problem in as many details as possible.

- Provide specific examples to demonstrate the steps.

- Describe the behavior you observed after following the steps and point out what exactly is the problem with that behavior. Explain which behavior you expected to see instead and why. If applicable, include screenshots which show you following the described steps and clearly demonstrate the problem.

- If you're reporting that the RobotController crashed, include the logfile with a stack trace of the crash.  [Example of good bug report with stack trace](https://github.com/ftctechnh/ftc_app/issues/224)

- If the problem wasn't triggered by a specific action, describe what you were doing before the problem happened and share more information using the guidelines below.

magneticflux-, 2016-10-28:
In general, an AAR file means that you are compiling an Android Library module. You must compile an app module in order to create an APK file. Be sure to specify the module to run a task on in Gradle.

LemonJustice, 2016-10-29:
I am doing Build APK from the Build menu.  There is no errors and I can open it, but I don't get a APK file.  Also, how do I specify the module in Gradle or specify the app module?

magneticflux-, 2016-10-29:
In the Gradle menu in Android Studio, expand the directory of the module you want to build, and then look for `packageRelease`. Execute that task, and then look under the `build\outputs\apk` directory for your unsigned APK.

LemonJustice, 2016-10-29:
Thank you!  That helped a lot!  TeamCode-release-unsigned.apk is the file that was generated.  Is this file I want to download to the RobotController?

EtherTyper, 2016-10-29:
Yes, it is.

magneticflux-, 2016-10-29:
I believe so, but the easier way to do this is by connecting the phone using ADB, syncing Gradle files in Android Studio, and running the app on the phone directly.

closed 2016-10-29 by LemonJustice