ftctechnh/ftc_app #252: Servo Range Issues

buggy213, 2016-10-30:
It's been a known issue for a while now that the servo ranges are limited due to the shortcomings of the MR Servo Controller (http://ftcforum.usfirst.org/archive/index.php/t-4461.html, from October 26 of last year). In short, the controller is unable to send the extreme ends of the PWM ranges that the HS-485HB expects (approx. 600us to 2400 us). The Servo Controller should theoretically be able to generate this; if you use the Core Device Discovery executable and go into the advanced settings for the servo controller, it is possible to extend the range of the controller. The implementation of this in the SDK is ServoEx (I think), but it is impossible to use because it you can't cast between the Servo and ServoEx interfaces when using the HS-485HB. Is there any way to fix this issue?

gearsincorg, 2016-10-30:
If the Discovery tool can expand the servo range, isn't this wider range "remembered" by the Servo controller?  Then the standard "Position" command issued to the servo will get expanded to fit the new range?

That aside, if you want to experiment with ServoEx, you just need to request that type of object when you do the hardware map.  You do need to use the new mapping format.

public ServoEx  myServo;

myServo = hardwareMap.get(ServoEx.class, "servo name");

buggy213, 2016-10-30:
The discovery tool can't make the extended range setting "stick", as far as I can tell. 

And as for your second point, in th javadoc for ServoEx, it says that you have to cast from the Servo interface - perhaps that needs to be updated?

Thanks.

buggy213, 2016-11-07:
Update: After contacting Modern Robotics Support Staff, they created a patch for the Core Device Discovery Tool that will now allow for the extended mode to "stick". Hooray!

calebsander, 2016-11-09:
@gearsincorg I can't get your code sample to work in release 2.35. `this.hardwareMap.getAll(ServoEx.class).size()` returns `0`, while `this.hardwareMap.getAll(Servo.class).size()` returns `10`. Since `ServoEx` does not extend `HardwareDevice`, it can't possibly be registered in the `HardwareMap`.

buggy213, 2016-11-09:
Yeah, the implementation of the ServoEx interface is really weird.

gearsincorg, 2016-11-13:
Sorry that I din't test this before posting.  I've used this approach on other device classes, so I assumed it would work on this one.  @buggy213 Was the patch only on the Discovery tool, or did it also update the Servo Controller firmware?

buggy213, 2016-11-13:
It was only for the discovery tool, so you still have to hook your computer
up to the servo controllers to get the full range of motion