ftctechnh/ftc_app #267: stuck in init()

MillburnHighSchoolRobotics, 2016-11-09:
When we initialize our Vuforia in init() (OpMode), many times we have an issue where we get stuck in init() due to the long time it takes. I believe this is an issue because of detectStuck method in OpModeManagerImpl. Looking for a way to disable this check. Thanks.

gearsincorg, 2016-11-09:
Hi

The detectStuck watchdog is a safety measure to ensure that an errant robot can be disabled remotely even if it's stuck in a loop.  It's not foolproof, but it's a reasonable measure.  

The current detectStuck timeout for inti() is 5 seconds.  This seemed reasonable when it was set, but it may be too short.  Does your init take longer than this?

For our reference, what are you setting up in Vuforia (number of targets or other actions etc) that may be extending the startup time?

If possible, it would also be helpful to know how long your Vuforia init can take, if left to it's own devices....     You could determine this by putting it in a dummy linearOpMode and just seeing how long it takes to run.   (When using linearOpMode, the app can only check the detectStuck when the Driver presses Stop.)

cmacfarl, 2016-11-09:
It is not possible to turn off the watchdog.

Can you please push a copy of your opmode to a team github repo?  I'd like to take a look at what you are doing?

TheKnownLawsOfAviation, 2016-11-13:
Hello,

Are you using LinearOpMode?, we used to use it and we were getting the same error. Adapting the code to just OpMode seemed to fix the bug.

MillburnHighSchoolRobotics, 2016-11-16:
Hi, we were able to fix our issue by starting Vuforia in a separate thread. Thanks

closed 2016-11-16 by MillburnHighSchoolRobotics