ftctechnh/ftc_app #268: Newly created PR guidelines are wrong.

EtherTyper, 2016-11-12:
> If you are looking at this button then you've pushed some changes to your team's fork of ftctechnh/ftc_app.

You make it seem like the viewer erred by creating a pull request to your repository. While it is true that if they properly create a pull request _**from**_ your repo _**to**_ their team repo, they won't encounter the PR template, I don't believe you should be discouraging the major method of getting upstream changes.

KingofGamesYami, 2016-11-12:
I don't believe they are discouraging it.  They're saying that if you _don't understand what you're doing_, you shouldn't be doing it.  And as you said, they won't encounter this message if they do things properly.

99.9% of the time, the viewer has erred by creating a pull request to this repository.  Just look at all the closed pull requests!

cmacfarl, 2016-11-12:
@KingofGamesYami, I prefer to characterize that sentence as, "If you don't understand what you are doing, here's an attempt to help explain it, and here are some resources for further reading, and oh by the way, in the friendliest manner possible, pushing that button probably isn't going to do what you think it will."  

@EtherTyper, respectfully, the vast majority of pull requests _are_ opened incorrectly.  The pull request template is simply designed to redirect attention to the contributing guidelines where the _intent_ is to educate, not discourage.   For those who have contributions where the intent is to indeed incorporate the change into the SDK, the contributing text notes that those are seen and evaluated on a case by case basis and the assumption is that submitter is familiar enough with git that the introductory text may be safely ignored.

closed 2016-11-12 by cmacfarl