ftctechnh/ftc_app #272: Weird Java Error

tf2hoovy, 2016-11-12:
I'm posting this here because the FTC Forum sucks.
My gradle build has throwing the following exception:

Error:Execution failed for task ':TeamCode:compileDebugJavaWithJavac'.
> java.lang.NullPointerException (no error message)

My code is at my github. Anyone have a solution?

KingofGamesYami, 2016-11-12:
You probably didn't initialize your hardware before attempting to reference it.  If this isn't the case, please point us to the file in question - I don't want to run through dozens of files trying to figure out which one is the problem.  You've also screwed up the placement of your files, though that probably won't cause any errors.

tf2hoovy, 2016-11-12:
As I said, it is not the code that is throwing this code; it is the Gradle Build.

My build.common.gradle: https://raw.githubusercontent.com/tf2hoovy/FTCBot/master/build.common.gradle

My TeamCode build.gradle: https://raw.githubusercontent.com/tf2hoovy/FTCBot/master/TeamCode/build.gradle

My TeamCdoe build.release.gradle: https://raw.githubusercontent.com/tf2hoovy/FTCBot/master/TeamCode/build.release.gradle

I'm pretty sure it has something to do with these but I don't know how Gradle works.

cmacfarl, 2016-11-12:
> com/tf2hoovy/FTCBot/master/build.common.gradle
> com/tf2hoovy/FTCBot/master/TeamCode/build.gradle
> com/tf2hoovy/FTCBot/master/TeamCode/build.release.gradle

I would suggest that you fork ftctechnh/ftc_app and then clone your fork
onto your local computer and work from that.  It's not entirely clear what
you've done with tf2hoovy/FTCBot and how it differs from ftctechnh/ftc_app

KingofGamesYami, 2016-11-12:
It appears you have modified it to add in the 'opencv' module which doesn't exist.  You should either remove the dependency on opencv (use the officially supported & already available Vuforia vision library instead) or include the opencv module.

Reading into things a little more, it appears you did not follow the recommended instructions for incorporating code from the FTC Vision repository.

https://github.com/lasarobotics/ftcvision/tree/9544b6081158e40f483768b6ed2191977f27a6f4#installing-into-first-official-app-recommended

gearsincorg, 2016-11-13:
[QUOTE]
I'm posting this here because the FTC Forum sucks.
[/QUOTE]

Most of the people who monitor this issues list, also monitor and respond to the forum, so saying that "It Sucks" is like slapping the faces of those you are asking for help...   Probably not the best way to get immediate action on your issue.

Also:  From someone who does their best to respond to people's issues...  can I mention that just saying "my project is broken, can you all fix it?" without providing ANY background as to what you did to get it there, is one of the reasons why it's hard to give good advice on the forums.

kev626, 2016-11-17:
You are likely not instantiating an object. Check for this issue. The other thing is you could be returning null on a function and trying to use the object. Check all functions that return null to find any bugs with them.

cmacfarl, 2016-11-17:
This doesn't appear to be a problem with the SDK itself.  For this reason, closing.  

If you are still having problems I'd encourage you to post your problem on the forums.   You will likely get better traction from the large number of smart and engaged people who monitor and participate in the forums.  And yes, I realize that the policies and software backing the forums may be less than optimal for some, but there are reasons for most decisions made.  Remember, FIRST is primarily a large number of volunteers, and a small number of staffers.  They all have the program and the student's interests front and center.

closed 2016-11-17 by cmacfarl

KevinEmery, 2016-11-18:
I just wanted to followup on this even though it is closed, I saw this today while building the repository (from the base code, no changes at all from master). The solution I found [here](http://stackoverflow.com/questions/39183674/java-lang-nullpointerexception-no-error-message) worked, which was to close the project, delete the `.gradle/` directory from the project directory (Note, NOT the `gradle/` directory), and then reopen the project in Android Studio.