ftctechnh/ftc_app #283: Modern Robotics Gyro sampling interval degrades rapidly when adding more I2C sensors.

trc492, 2016-12-03:
The issue is detailed in this forum thread.
http://ftcforum.usfirst.org/showthread.php?7529-Odd-problem-with-Core-Device-Interface-(seems-to-impact-both-Color-Sensor-and-ODS)
I have written a simple opmode to test the issue and noticed that when we have 4 I2C sensors: 1 MR gyro, 2 MR color sensors and 1 MR range sensor, the gyro heading data doesn't change for approx. 300 msec. This performance is unacceptable for doing close loop control. I can't explain why the sampling interval is so bad considering the I2C bus is running at 100 kHz (approx. 100 kbps). because each sensor would have to transmit 1 kbyte for each bus transaction in order to have this kind of bad performance. So I suspect it is an I2C software architectural issue. Please look into it. Our robot can't turn accurately as a result. It often oscillates indefinitely. As a quick remedy, if you could provide an API to enable/disable an I2C device reader, that will help because most of the time, we will be using just one I2C device at a time. So we can turn off other I2C devices. For example, when the robot is turning, it is unlikely we would access the color sensor that detects the beacon 

color, or the color sensor to detect a line.
[TestCode.zip](https://github.com/ftctechnh/ftc_app/files/628761/TestCode.zip)

trc492, 2016-12-03:
Sorry wrong test code. This is a simpler version that doesn't depend on our library.
[code]
package samples;

import android.util.Log;

import com.qualcomm.hardware.modernrobotics.ModernRoboticsI2cGyro;
import com.qualcomm.robotcore.eventloop.opmode.Disabled;
import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;
import com.qualcomm.robotcore.eventloop.opmode.TeleOp;
import com.qualcomm.robotcore.hardware.DcMotor;

@TeleOp(name="Test: Simple Sensor Sample Time", group="3543TestSamples")
@Disabled
public class SensorSampleTimeTest extends LinearOpMode
{
    private enum SensorType
    {
        DRIVEBASE_ENCODERS,
        GYRO
    }

    private static final String TAG = "TrcDbg";
    private static final double DRIVE_POWER = 0.2;
    private static final double TURN_POWER = 0.5;
    private static SensorType sensorType = SensorType.GYRO;
    private static final DcMotor.Direction LEFTWHEEL_DIRECTION = DcMotor.Direction.FORWARD;
    private static final DcMotor.Direction RIGHTWHEEL_DIRECTION = DcMotor.Direction.REVERSE;

    private DcMotor lfWheel;
    private DcMotor rfWheel;
    private DcMotor lrWheel;
    private DcMotor rrWheel;
    private ModernRoboticsI2cGyro gyro;

    public void runOpMode()
    {
        initRobot();

        waitForStart();

        long minLoopInterval = Long.MAX_VALUE;
        long maxLoopInterval = Long.MIN_VALUE;
        long loopCount = 0;
        long prevLoopTime = 0;

        long minSampleInterval = Long.MAX_VALUE;
        long maxSampleInterval = Long.MIN_VALUE;
        long sampleCount = 0;
        long prevSampleTime;

        long startTime = System.nanoTime();
        prevSampleTime = startTime;
        int prevSample = getSensorValue();

        while (opModeIsActive())
        {
            long currTime = System.nanoTime();
            int currSample = getSensorValue();
            if (prevLoopTime != 0)
            {
                long loopInterval = currTime - prevLoopTime;

                if (currSample != prevSample)
                {
                    long sampleTime = currTime - prevSampleTime;
                    sampleCount++;
                    prevSample = currSample;
                    prevSampleTime = currTime;
                    if (sampleTime < minSampleInterval)
[... 90 lines cut ...]
                break;

            case GYRO:
                //
                // Turning right and checking gyro.
                //
                lfWheel.setPower(TURN_POWER);
                lrWheel.setPower(TURN_POWER);
                rfWheel.setPower(-TURN_POWER);
                rrWheel.setPower(-TURN_POWER);
                Log.i(TAG, prefix + String.format("heading=%d", -gyro.getIntegratedZValue()));
                break;
        }
    }   //runRobot

    private void stopRobot()
    {
        lfWheel.setPower(0.0);
        lrWheel.setPower(0.0);
        rfWheel.setPower(0.0);
        rrWheel.setPower(0.0);
    }   //stopRobot

}   //class SensorSampleTimeTest
[/code]

gearsincorg, 2016-12-06:
Note:  Several of the developer team are looking into this.  Our goal is to provide the best possible performance.

However, these matters take time to analyse and resolve, so please don't read the lack of an immediate solution as a lack of interest.  Far from it.

haithcoc, 2016-12-09:
We have had the same issue.  We have one MR I2C gyro and one MR I2C color sensor.
With both connected, the time between gyro heading changes was ~100ms.  At a turn rate of 100 deg/sec we would see 100ms with gyro reporting the same heading value, followed by a 10 deg jump and 100ms of the new value.
When we remove the color sensor from the config file, The time between updates dropped to ~35ms.
We also tried deregister of the port ready callback for the color sensor via:
colorSensor.getI2cController().deregisterForPortReadyCallback(colorSensor.getPort());
and the rate of gyro sensor heading changes went from 100ms to 30-40ms.
When we subsequently called 
colorSensor.getI2cController().registerForI2cPortReadyCallback(colorSensor, colorSensor.getPort());
we can turn on the color sensor LED, but were not getting values to find the line.

Is there any activity on this issue?

gearsincorg, 2016-12-09:
I've been able to replicate the issues and I'm working directly with the originator of the code to address the cause of the delays.  At this point we are characterizing the system and trying to establish where the bottleneck is. 
We've also seen an unexpected dependency on the number of other CORE devices in the configuration file (regardless of access) so we will also explore that issue.

When we have a possible culprit/resolution I'll pass it on.   If there is an immediate fix I'll lobby for an update to the beta branch.

trc492, 2016-12-09:
How about an immediate workaround by providing an API to disable/enable an I2C device? IMHO, this is a good long term API anyway, not just a temp fix.

SpencerSharp, 2016-12-20:
How's resolving this issue going? It's really important to my team so I was just wondering if it seems solvable or not, and what time frame you're looking at on it.

gearsincorg, 2016-12-20:
Refer to this post on the general forum:

http://ftcforum.usfirst.org/showthread.php?7724-Gyro-Read-Latency-Development-team-update

cmacfarl, 2016-12-20:
The beta branch containing 2.6 fixes this issue.

closed 2016-12-20 by cmacfarl

trc492, 2016-12-20:
What about the other related performance issues? Beta 2.61 only fixes the gyro performance but other sensor performance issues still exist. Should I open a separate ticket?