ftctechnh/ftc_app #287: DC motor running too slow in RUN_USING_ENCODER run mode

AlecHub, 2016-12-12:
When we select `RUN_WITHOUT_ENCODER` run mode and `setPower(1.0)`, our unloaded NeveRest 20 DC motor runs at 2800 encoder counts per second (CPS).  

Therefore, when we select `RUN_USING_ENCODER` run mode, `setMaxSpeed(2800)`, and `setPower(1.0)`, we expect the same motor to run at a constant 2240 CPS.  Instead, the motor runs at roughly 1800 CPS.

gearsincorg, 2016-12-12:
This is another "feature" of the Modern Robotics motor controller.
Their firmware chooses to enforce a certain amount of overhead on any constant speed mode (RUN_USING_ENCODER and RUN_TO_POSITION).  
This means that their controller will not actually ever try to reach full speed in either of these modes.

They claim it is a feature, I have been fighting for many years that it is an annoying, and undesired constraint.  I feel that they should let the programmer decide how much overhead they want, based on the specific motor/application. 

That said, if you have a motor that naturally has a lower CPS, then you can bump up the MaxSpeed() and simply request a lower power...  This gets around the forced overhead.

eg: in RUN_TO_POSITION, the MR controller limits (clips) the actual top speed to about 65% of MaxSpeed.  So if you leave Max Speed set at 4000 CPS, and never request more than 0.65 power, you will effectively approach the top speed of your motor.  

You do need to monitor performance in the field to ensure that you aren't demanding too much speed.  If you find that the motor is slow at responding to a reduction in requested power, you may have exceeded the motor's capabilities.

Note that this is NOT a FTC_SDK issue, it's a long standing Hitechnic/ModernRobotics issue.

AlecHub, 2016-12-12:
Not all applications require the 20% CPS headroom that the motor controller firmware and/or the FTC SDK imposes with the `RUN_USING_ENCODER` run mode.  

Therefore, we request a new run mode, `RUN_AT_CONSTANT_SPEED`, and a new method, `DcMotor.setSpeed(int encoderTicksPerSecond)`.  With this new run mode, the motor is to do its best to run at the targeted `encoderTicksPerSecond` constant speed.

Users must be careful to select a `encoderTicksPerSecond` value that is achievable by the particular DC motor for the particular application, failing which users understand that the targeted speed, or constant speed, or both, will not be achieved.

The new run mode likely requires changes to the motor controller firmware, but seems trivial to implement given that it is a simplification of existing functionality.

gearsincorg, 2016-12-12:
Again... The FTC_SDK is not imposing any limitations, it's simply accessing the Motor Controller API.  It cannot ask for something that the firmware will not provide.

You should ask Modern Robotics for this "trivial" change directly.    

Here is the closest you will get to a specification for the Hitechnic/Modern Robotics motor controller API.
http://www.hitechnic.com/blog/wp-content/uploads/HiTechnic-Motor-Controller-Specification.pdf
Note: it's very old.

AlecHub, 2016-12-12:
I am requesting FIRST to request Modern Robotics for this feature on behalf of the teams since ultimately a new feature in the firmware is of no use if FIRST is not included to expose the new feature in the SDK.

magneticflux-, 2016-12-12:
Can't a software workaround be implemented? Is the limit exactly known, or is more testing required?

AlecHub, 2016-12-13:
> Environment:
> * NeveRest 20 Gearmotor (am-3102)
> * Core Motor Controller: rev 2.0 (configured for AndyMark)
> . . .

Furthermore, if we put the motor controller in Matrix mode, we get 2800 CPS, and in Tetrix mode we get 2360 CPS.  

In summary, the following are our CPS readings when we call `setMaxSpeed(2800)`, and `setPower(1.0)`:

 | Motor controller mode | CPS | 
 |-----------------------|----------------------------------------|
 | AndyMark              | 1800                                   | 
 | Tetrix                | 2360                                   | 
 | Matrix                | 2800                                   |

AlecHub, 2016-12-13:
> Can't a software workaround be implemented? Is the limit exactly known, or is more testing required?

The [documentation on setMaxSpeed()](http://ftctechnh.github.io/ftc_app/doc/javadoc/com/qualcomm/robotcore/hardware/DcMotor.html#setMaxSpeed(int)) says:

> _When the motor is running in one of the PID modes the value set using the `setPower()` method is indicative of a desired motor velocity rather than a raw power level. In those modes, the `setMaxSpeed()` method provides the interpretation of the speed to which a value of 1.0 passed to `setPower()` should correspond._

There is no mention in the documentation as to what speeds `setPower()` maps to nor that `setMaxSpeed()` yields different mappings depending on the motor controller mode.  

Thus the need for a new   `RUN_AT_CONSTANT_SPEED` run mode which, by design, will have none of the voodoo of the `RUN_USING_ENCODER` run mode.

AlecHub, 2016-12-22:
> There is no mention in the documentation as to what speeds  `setPower()`  maps to nor that  `setMaxSpeed()`  yields different mappings depending on the motor controller mode.

**Correction:** The documentation says `setPower(1.0)` maps to the `encoderTicksPerSecond` value passed to `setMaxSpeed()`.  So there is either a bug in the documentation or a bug in the SDK.

gearsincorg, 2016-12-22:
You guys are getting hung up on the concept that the SDK can issue commands to the motor controllers that indicate the actual Encoder Counts Per second that are required.

The problem is that it can't.  

The only API that the motor controller offers is a +/- 0 to 100 integer range.
This is interpreted as 0 to full power, or zero to full speed.
Since the new FTC_SDK was doing everything in the unity range, this was just re-scaled to +/- 0.0 to 1.0

And here is the big issue...  Hitechnic and Modern Robotics (yes Alec, I know there is old H/W and new H/W  ) are free to interpret 100% speed any way they want.  All the artificial speed limiting that existed in their original hardware was migrated to the new hardware, so that's why my history is relevant.  

I have spent YEARS trying to get Hitechnic and Modern Robotics to truly map 0-100 to the full 4000 CPS that they state in their spec, but the reality is they don't want to, and nothing will change their mind.  

Then when more motors came out (like all of the Andymark ones) a whole part of the 0-100 scale became unusable because the Andymark motors could only generate 2600 CVS.  So any speed request over 0.65 was unobtainable.  Even less with Run To Position.

So given that, FIRST (though their SDK) did attempt to initiate a "Software work around" that would enable users to still specify a 0-1 power/speed range, that would map to a lower range of Encoder Counts per second.  The intent was that by specifying what your motor's real max speed was, the SDK would simply translate your 0-1 range to the corrosponding motor controllers range.. eg: 0-0.65 for Neverest motors.

This was no silver bullet, it was just a convenience.  
So, if you read the current docs now in this context...

"setPower(1.0) maps to the encoderTicksPerSecond value passed to setMaxSpeed()"

It's an exact description of what's being done.   If this doesn't make sense, look at it a bit harder.  Run your test data through the math.

If you think this should be better, or smarter or more logical..  I agree, but you are barking up the wrong tree here.  Without an updated, public API from MR, not much can change.

As developers, we've been responsive to new hardware and updated interfaces.  If a new interface became available on the MR controllers, I'd be the first one to push for using it.

But if YOU (or the WE you refer to) want MR to change , YOU need to beat on THEIR door, and get THEM to publish a smarter interface.

AlecHub, 2016-12-22:
Phil, according to the **Modoern Robotics documentation**:

> **Run at constant speed:** This mode controls the speed of the motor such that it maintains an average speed. If the motor is set to 85% power in this mode, the controller will monitor the encoder to give an appropriate power to the motor such that the motor rotates at 85% of its maximum average speed ...

What this means is that if I have a AndyMark 20/40/60 motor, I simply configure my Core Motor Controller with the AndyMark PID parameters, then enable RUN_USING_ENCODER run mode in my opmode, and then if I am somehow able to pass POWER=100 to the firmware, then the firmware will autimatically calculate 85% of maximum average speed of my AndyMark 20/40/60 motor and run it at that speed.

Likewise if I have a TETRIX motor, all I have to do is configure the motor contorller with the Tetrix PID parameters and pass POWER=100 to the firmware and the modern robotics firmware will automatically calculate 85% of maximum average speed of my Tetrix motor and run it at that speed.

If I want my AndyMark or Tetrix motor to run at a particular RPM, I can calcualte the POWER value I need to pass to the firmware and the firmware will run my motor at the RPM I need.

If the Core Motor Controller doesn't work per the Modern Robotics documention, I can contact Modern Robotics and they will fix it.

**The problem is** the SDK takes the setPower() and setMaxSpeed() values and and then calculates a POWER value to send to the firmware.  Clearly teams are in no position to contact Modern Robotics if teams don't know what POWER value the SDK is sends the firmware.  Moreover, the POWER value that the SDK is sending to the firmware is not the value that the SDK documentation says it should be sending to the firmware.

cmacfarl, 2016-12-22:
> *The problem is* the SDK takes the setPower() and setMaxSpeed() values
and and then calculates a POWER value to send to the firmware.

If your opmode never calls setMaxSpeed() then the calculation you are
referring to devolves to

  power = power;

It's only if you redefine what maxSpeed is that the power you send to the
controller will be scaled to fit the maxSpeed requested.  So you can know
exactly what power value the SDK is sending to the controller.  Just don't
use setMaxSpeed if that's what you want.

Phil's explanation of what setMaxSpeed's purpose is above should be baked
more tightly into the SDK documentation.  As if I'm just reading the
javadoc I'm thinking, ok, what's the point?

However, my read of all this is if you configure your controllers via the
MR utility the way MR specifies (and this needs better, more visible
documentation for the community), then I believe your motors should behave
the way you expect just using setPower() alone.

AlecHub, 2016-12-22:
> If your opmode never calls setMaxSpeed() then the calculation you are referring to devolves to
>
>   power = power;
>
> It's only if you redefine what maxSpeed is that the power you send to the controller [...]

@cmacfarl,  so what you are saying is the `encoderTicksPerSecond` value passed to `setMaxSpeed()` has nothing whatsoever to do with any **_actual_** maximum encoder ticks per second of any **_actual_** motor.  

If my motor happens to have a maximum no-load encoder ticks per second of say 9000 and I call setMaxSpeed(4000), then setPower(1.0) will send POWER=100 to the motor controller firmware, and setPower(0.5) will send POWER=50 to the firmware.  Is this correct?

> Phil's explanation of what setMaxSpeed's purpose is above should be baked more tightly into the SDK documentation.  As if I'm just reading the javadoc I'm thinking, ok, what's the point?

What Phil is saying about `setMaxSpeed()` (both here and in the FTC forum) is different than what you are saying (if I am correct on what you are saying).

cmacfarl, 2016-12-23:
> so what you are saying is the encoderTicksPerSecond value
> passed to setMaxSpeed() has nothing whatsoever to do with any
> actual maximum encoder ticks per second of any actual motor.

In a word, maybe, yes?

This is perhaps a case where a pictu^H^H^H^H^H, software speaks louder
than words.

setPower() eventually does this...

  double defMaxSpeed = getDefaultMaxMotorSpeed(motor);
  power = Math.signum(power) * Range.scale(Math.abs(power), 0,
apiPowerMax, 0, motors[motor].maxSpeed / defMaxSpeed);

motors[motor].maxSpeed will be equal to defMaxSpeed if you don't call
setMaxSpeed and hence, ignoring all the sign cruft,

  power = Range.scale(power, 0, apiPowerMax, 0, 1);

And while getDefaultMaxMotorSpeed() might imply that there's a way to
change the default, there is not.  It defaults
to tetrix values.

Part of the problem here is that there's no way for a controller to
query the motor and figure out what it is, and perhaps another
part of the problem is that there's no way to tell the SDK that a
particular port has a particular motor.

Which therefore devolves into, it's better to configure the
controller's firmware via the MR utility, don't call setMaxSpeed, and
let the controller do it's thing with the power value passed
unmodified to the controller.   Assuming of course you don't mix
motors on a single controller.

AlecHub, 2016-12-24:
> setPower() eventually does this...
>
>  double defMaxSpeed = getDefaultMaxMotorSpeed(motor);
>  power = Math.signum(power) * Range.scale(Math.abs(power), 0, apiPowerMax, 0, motors[motor].maxSpeed / [...]

Thanks Craig, but you and Phil do understand that, should the motor controller firmware not perform per the Modern Robotics specs, the code you've provided is insufficient to take to Modern Robotics to open an issue with them, regardless of whether or not you have permission to publish that portion of SDK code.

As long as teams do not have direct control of the POWER value that is sent to the motor controller firmware, teams are in no position to open issues against Modern Robotics.

> Part of the problem here is that there's no way for a controller to query the motor and figure out what it is, and perhaps another part of the problem is that there's no way to tell the SDK that a particular port has a particular motor.

I've requested that the Robot Controller App allow the type of motor to be selectable for each port of a motor controller (#297).  If the type of motor is configurable with the Core Device Discovery App, I'm sure Modern Robotics will graciously give FTC the APIs and the libraries needed to implement this feature in the Robot Controller App.

rgatkinson, 2016-12-24:
There are more things in heaven and earth, Horatio, 
Than are dreamt of in your philosophy.

gearsincorg, 2016-12-24:
> Thanks Craig, but you and Phil do understand that, should the motor controller firmware not perform per the Modern Robotics specs, the code you've provided is insufficient to take to Modern Robotics to open an issue with them

Seriously Alec ?    I don't see how we can get more "sufficient" than: power = power.

It seems to me that you are purposefully trying to ignore the obvious here.
Here it is again, as clearly as I can make it.   No code, no fancy formula, just one basic fact.

Here's how you know EXACTLY what's being sent to the motor controller.

1) DO NOT call setMaxSpeed()  .  Never, not once, ever, at all, in your opmode.  Don't even consider it.

2) Make a call to setPower(), passing a number between -1 and +1

3) The SDK will multiply your value by 100 and send it to the motor controller.
     (it's multiplied by 100 since the SDK uses +/- 1 and the controller uses +/- 100)
     In terms of "requested power", it's a direct 1 to 1 relationship...

That's it.  Done...   not super complicated...
You wanted to know how to send specific values to the motor controller.. Now you know.
Don't call setMaxSpeed(), do call setPower()

So... your problem with "insufficient" information is solved... right?

LBYPatrick, 2017-12-03:
You can actually write a function that uses "getCurrentPosition()" and "setPower()" to achieve the same purpose just like writing a DriveTrain control class. By the way, although our team s**ks at autonomous, we did make **safe zone parking** and **placing glyph in the Cryptobox** working perfectly without any sensor (even include encoders).

closed 2017-12-03 by cmacfarl