ftctechnh/ftc_app #289: com folder missing?

ijsalamon, 2016-12-13:
I don't knwo if I can't find it or if its not there, but whenever I grad this folder, the path com.qualcomm.ftcrobotcontroller is missing which limits us from getting the robot to work. If anyone has any idea where it is or how to get it, it would be a major help to me.

Thanks,

Ian S

EtherTyper, 2016-12-13:
I believe that package is stored in the `jar` file [`libs/RobotCore-release-sources.jar`](https://github.com/ftctechnh/ftc_app/blob/master/libs/RobotCore-release-sources.jar)

magneticflux-, 2016-12-13:
@ijsalamon The fact that the path is not resolved indicates that you cloned the repository incorrectly. Use a Git client to clone the repository or fork it online using Github, and put your code in the TeamCode Gradle module.

cmacfarl, 2017-12-04:
Housekeeping.

closed 2017-12-04 by cmacfarl