ftctechnh/ftc_app #307: ConceptI2cAddressChange crashes writing error message (and doesn't work)

slylockfox, 2017-01-16:
ConceptI2cAddressChange gives the following exception:

`Event loop threw an exception
java.util.IllegalFormatConversionException: %x can't format com.qualcomm.robotcore.hardware.I2cAddr arguments
C_CONCURRENT freed 1962K, 45% free 6216K/11164K, paused 3ms+4ms, total 58ms
    at java.util.Formatter.badArgumentType(Formatter.java:1489)
    at java.util.Formatter.transformFromInteger(Formatter.java:1689)
    at java.util.Formatter.transform(Formatter.java:1461)
    at java.util.Formatter.doFormat(Formatter.java:1081)
    at java.util.Formatter.format(Formatter.java:1042)
    at java.util.Formatter.format(Formatter.java:1011)
    at java.lang.String.format(String.java:1999)
    at java.lang.String.format(String.java:1973)
    at org.firstinspires.ftc.teamcode.UselessChangeIRAddress.runOpMode(UselessChangeIRAddress.java:142)
    at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:303)
    at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:513)
    at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:298)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
    at java.lang.Thread.run(Thread.java:841)
`
I assume it is trying to report that it cannot read bytes as expected, so another more serious issue is that the sample doesn't change addresses.  I have tried to change the address of an IR Seeker using Core Device Discovery, but the address change does not persist after turning the robot off and back on, so I attempted to use this sample.  I made no changes, other than copying it to Team Code.

SamPhilip, 2017-01-16:
I used the Core Device discovery for changing the I2C address of my three colour sensors and then in the program I use:
<code>beaconFinder.setI2cAddress(I2cAddr.create8bit(0x66));</code>
I used a colour sensor but I think it will work with the IR seeker too.

Just make sure that the I2C address of your sensor in your program is the same as what you set it with the Core Device Discovery. Also when using the core device discovery, make sure you are plugging in one I2C sensor into a Core Device Interface Module at a time when changing I2C address (you shouldn't have to do that in . If you try to change the I2c address with multiple sensors of the same address, you will end up changing all of them. 

Hope this helps and good luck.

calebsander, 2017-01-16:
I think the crash comes from this line: `telemetry.addData("I2cAddressChange", String.format("Looping too long with no change, probably have the wrong address. Current address: 0x%02x", currentAddress));`. `currentAddress` is an instance of `I2cAddr`, but was probably a numerical value when the code was first written (before version 2.0). It should probably read `currentAddress.get7Bit()` instead. A similar problem can be seen on lines 143, 175, and 177.

slylockfox, 2017-01-30:
I have changed addresses of color sensors successfully, but my issue is specifically with the IR seeker sensor.  This forum post discusses the same issue I'm having, except with the range sensor. It seems like maybe only the color sensor can have its address changed permanently.  IR seekers are obviously not critical for this year's game, but are still very useful for demo robots.
http://ftcforum.usfirst.org/showthread.php?7511-Problem-using-2-of-ModernRoboticsI2cRangeSensor-and-or-ColorSensor

slylockfox, 2017-02-21:
@calebsander  do you mean get8Bit() ?  get8Bit() is used elsewhere in the code.

After fixing the string formatting, it of course throws an exception on the next line, which seems intentional, to make an early exit
        `hardwareMap.irSeekerSensor.get(String.format("Looping too long with no change, probably have the wrong address. Current address: 0x%02x", currentAddress.get8Bit()));`

Should it be using 7Bit everywhere instead of 8Bit?  For example, should this line
`public static final I2cAddr IR_SEEKER_V3_ORIGINAL_ADDRESS = I2cAddr.create8bit(0x38);`
be this instead
`public static final I2cAddr IR_SEEKER_V3_ORIGINAL_ADDRESS = I2cAddr.create7bit(0x38);`

calebsander, 2017-02-21:
It doesn't really matter whether you write I2C addresses as 7-bit or 8-bit, as long as you are consistent. Modern Robotics uses 8-bit ones, but almost every other manufacturer uses 7-bit addresses because addresses only contain 7 bits. The 8-bit address is just twice the 7-bit address (meaning it can take on even values from 0 to 254 rather than any value from 0 to 127). It probably makes sense to use 8-bit addresses everywhere for consistency in the code. I suppose I am just used to using 7-bit addresses for Adafruit sensors.

slylockfox, 2017-02-22:
Thanks for the quick reply!  I tried changing the calls to 7Bit everywhere, but same result

calebsander, 2017-02-22:
Hmm, the line it's crashing on makes no sense to me. It creates a message for the user and then tries to use that string as the name of a device in the hardware map, which is obviously not going to work. `hardwareMap.get()` will throw an exception if it can't find the device with that name. That code is only run if the address change fails, so perhaps it is a convoluted way of throwing an exception in order to quit the program. I would recommend using the Core Device Discovery utility instead if you actually want to change the I2C address of a device.

slylockfox, 2017-02-22:
That was my thought as well, about the strange hardwareMap line, and I changed it.  The reason I'm messing with this code is that Core Device Discovery doesn't seem to be able to permanently change the address of the IR Seeker.  The IR Seeker's address always reverts to factory after a power cycle.

slylockfox, 2017-02-22:
I got it working.  The main issue was this line; the firmware version of mine is 0x14
`public static final byte IR_SEEKER_V3_FIRMWARE_REV = 0x14; // was 0x12`

Now I can run this (with one IR Seeker plugged in), change the address, plug in the other, and have two IR Seekers.  The sensor still resets to factory after a power cycle.

calebsander, 2017-04-13:
Looks as though this was fixed in v3.0.0: https://github.com/ftctechnh/ftc_app/commit/8e3f545428f5ffaeb2bd515d2c2aa3b172492f1d#diff-9b8d29ff50d46272488d4abc77b3371eR142