ftctechnh/ftc_app #308: Range Sensor Consistency with other Modern Robotics Sensor Libraries

SamPhilip, 2017-01-16:
I noticed that the range sensor's library is different from that of the other modernRobotics sensors. Normally what I would do is call:
<code>public ColorSensor colorSensorL;</code>
<code>public void runOpMode() {  </code>
<code> colorSensorL = new ModernRoboticsI2cColorSensor(cdim, 0); </code>
<code>// ...</code>
<code>}</code>
With the range sensor, the constructor is something like this:
<code> new ModernRoboticsI2cRangeSensor(I2cDeviceSynch deviceClient)</code>

Could someone explain why the range sensor is like this?

Does it have to do with being a combo ultrasonic/optical distance sensor?

calebsander, 2017-01-16:
The range sensor and compass sensor drivers were written at the start of this season, after the synchronous I2C library was added to the SDK, whereas the color sensor driver has been around since the previous year. I believe that because the synchronous format is much easier to use to write a sensor driver, they decided to go with it for these new drivers. Since the old ones already worked, I don't think there was any reason to change them.
I don't see why you need to manually construct the sensor objects, though. Can't you just register a color sensor or a range sensor on I2C port 0 in the config file?

SamPhilip, 2017-01-16:
In my opinion, they should make it so you can manually construct the sensor objects, that way, I can read and write commands directly through the sensor via the core device interface, especially with I2c sensors. My Gyro sensor wouldn't calibrate properly when plugged into logical port 2, so I plugged it in to port 4 and changed the port number in the code and it works fine. Honestly, it is more convenient to change it in the code than to reconfigure the robot, because if you reference a hardware device using <code>hardwareMap</code> then the program would stop working if the device didn't exist (I plug the sensor into a different port).

I understand that the range sensor is a fairly new sensor so they wrote the sensor driver differently. In that case, how would I use the constructor for the MR Range Sensor. Is there any backward compatibility with older sensor drivers.

calebsander, 2017-01-16:
The I2C ports are actually all connected to the same bus, so that doesn't really make sense. If you want to change the configuration from Androi Studio alone, you can also just use a [resource config](http://ftcforum.usfirst.org/showthread.php?6841-Robot-Configs-as-resources-and-the-like). If you are bent on constructing it manually, I believe this should work:
````java
final DeviceInterfaceModule dim = (DeviceInterfaceModule)this.hardwareMap.get("dim");
final I2cDevice rangeDevice = new I2cDeviceImpl(dim, 1 /*port*/);
final I2cDeviceSynch rangeDeviceSynch = new I2cDeviceSynchImpl(rangeDevice, true);
this.rangeSensor = new ModernRoboticsI2cRangeSensor(rangeDeviceSynch);
````

SamPhilip, 2017-01-16:
Thanks a lot, I'm sure it will help.

SamPhilip, 2017-01-16:
I think I figured it out for now

SamPhilip, 2017-01-16:
They should provide more documentation for this sensor

closed 2017-01-16 by SamPhilip

EtherTyper, 2017-01-16:
@SamPhillip They [welcome contributions](https://github.com/ftctechnh/ftc_app/blob/master/.github/CONTRIBUTING.md).