ftctechnh/ftc_app #310: ModernRoboticsI2cColorSensor makes incorrect assumptions about the starting state of the sensor LED

guineawheek, 2017-01-17:
Assuming one has a `ModernRoboticsI2cColorSensor` instance called `colorSensor` If one runs `colorSensor.enableLed(true)` without calling `colorSensor.enableLed(false)` first chances are color sensor's LED won't actually turn on. 

This is because of a piece of optimization in the `enableLed` function:

```
  if (lastCommand == command) { // switching is expensive
    return;
  }
```

This would be fine if it weren't for the minor detail the default value of `lastCommand` is always `COMMAND_ACTIVE_LED`, so when a call to turn the LED on happens the value of `command` is also set to `COMMAND_ACTIVE_LED`, and the function will return at this snippet instead of writing the command to the I2c write buffer. Judging from the comment where the `lastCommand` field is declared, `lastCommand` is set to `COMMAND_ACTIVE_LED` because the NXT sensors start with the LED always on. However, Modern Robotics color sensors can start with either the LED on or off, making the assumption that they always turn on to be invalid. 

The fix seems pretty simple: initialize the `lastCommand` field to have a value of neither `COMMAND_ACTIVE_LED` nor `COMMAND_PASSIVE_LED`, like -1.

dvick, 2017-03-10:
I just noticed this myself the other day, and came here to see if it had been fixed or if someone made an issue on it. Glad to see someone has started an issue on it, but it looks like it's been ignored thus far, unless it was just fixed without updating the issue; I haven't checked the latest commit yet, but I'm guessing it hasn't been.

> also, why are there comments about NXT sensors in an i2c sensor class?!

My guess would be that someone simply copied and pasted that line from the class for the NXT sensor, and didn't think about it any further, although I haven't checked that that same line is in the NXT class, to be fair.

NoahAndrews, 2020-09-03:
This got fixed a long time ago.

guineawheek, 2020-09-06:
cheers

closed 2020-09-06 by guineawheek