ftctechnh/ftc_app #311: something wrong with WIFI direct on android 6.0

heyuhang0, 2017-01-19:
I found that I can't use "change wifi channel" in the robot controller app on Android 6.0.When I try to change WIFI channel,it just closes WIFI but never switch it on. And the app showes that there is an error with connection.I tried it on two different phones,but there was same problem.

EtherTyper, 2017-01-19:
Android 6.0 isn't approved, according to page 35 of [Game Manual Part 1](https://firstinspiresst01.blob.core.windows.net/ftc/game-manual-part-1.pdf). Only 6.0.1+.

EtherTyper, 2017-01-19:
To diagnose the issue, it would also help to know what model of phone you use.

Mark1082, 2017-01-19:
Not all phones support that app.  On the inspection sheet it says "The FTC Wi-Fi Direct Channel Changing App is installed on the Robot Controller (for ZTE Speed devices only).
Robot Controller is set to the correct Wi-Fi Direct channel (if required by the tournament, ZTE
Speed, Motorola Moto G 2nd Generation, Motorola Moto G 3rd Generation, and Motorola Moto G4
Play only)."

Depending on the phone you are using, it might not be possible to change the wifi channel.  If you are on the Google Nexus or Samsung Galaxy it might not be possible.

closed 2017-01-22 by heyuhang0

carolineachee, 2017-01-23:
We have a Motorola Moto which was sent by FIRST and we are not able to download the "FTC Wi-Fi Direct Channel" program. Our league at times requires that we change the channel. 

The attached are the images of the phone revision and the error when the download stops. 

[image]
[image]

You help is greatly appreciated.
Thanks!

twlum, 2017-01-23:
I don't believe Moro can change channels. Only the ZTE Speed can download
and install this app.

NOTICE: The information contained in this e-mail may be confidential and is
> intended solely for the use of the named addressee. Access, copying or
> re-use of the e-mail or any information contained herein by any other
> person is not authorized. If you are not the intended recipient please
> notify us immediately by returning the e-mail to the originator.
>

>

<https://cloud.githubusercontent.com/assets/7811538/22221240/1f838a56-e180-11e6-93d2-90d0cffd948b.JPG>
[image: img_3293]
<https://cloud.githubusercontent.com/assets/7811538/22221241/1f8a4c38-e180-11e6-86ec-8f2533e21aa7.JPG>

KingofGamesYami, 2017-01-24:
Our team used the Moto G4 Plays, we were able to change channels directly from the Robot Controller App.

gearsincorg, 2017-01-24:
As stated earlier....  

If you are running on a Moto G phone, there is an option on the Robot Controller's phone (RC) to change channels.

Open the RC app, and use the three dots menu to open the Settings screen.

The first item on the list is the channel changer.

Make a selection and then hit the button the lock it in.

No need to make a change on the Driver Station... it follows the RC.