ftctechnh/ftc_app #314: Continuous Servos cannot be setup in FTC RC

jacobbednarski, 2017-01-27:
When I declare a 360 servo in android studio so I can get it to function like a motor for a lift on our robot I have button a move the servo to set positon 1 and y to set position 0 and then saying else .5 which I was told should stop the servo. The servo will still act as if it isn't a continuous rotation servo when I leave it as a servo in FTC RC but when I change it to a continuous servo it throws back an error saying I'm not declaring that servo when hitting initialize. Is there any one who had a similar problem or has some sample code for continuous rotation servos on how to set them up different from regular servos in android studio? Thanks

gearsincorg, 2017-01-27:
From your description I'm not sure if you are being consistent with the servo declarations.

You can use your existing servo code with a CR servo, if you use the values you mention... 0, 0.5, 1.0

However, if you want the code to know that it's a CR servo (and set it off when the opmode ends)
you need to change the servo type in your code AND in the robot config.

So you need...  
1) to set the servo as CR in the robot config page. (done)
2) and you need to declare and use the servo as CR in the code:

```
    // Declare a Cont. Rot servo
    CRServo arm;

    // Connect CR servo to correct servo on robot
    arm = hardwareMap.crservo.get("arm servo");

    //  now use buttons to change arm motor.
    if (gamepad1.x)
        arm.setPower(1)
    else if (gamepad1.a)
        arm.setPower(-1)
    else
        arm.setPower(0)
```

Note: Once you have a CR servo in code, you need to call setPower() on it, rather than setPosition()

If you declare the servo as CR in the code, but do not change the servo type on the phone, then the code will not "SEE" the servo in the configuration.  This is the cause of the init error.

jacobbednarski, 2017-01-27:
That makes more sense. Using CRServo instead of servo. That helps me a lot. Thank you!

closed 2017-01-27 by jacobbednarski