ftctechnh/ftc_app #325: Multiple Range Sensors

FrostDragon13, 2017-02-14:
Our FTC team this year were planning on using two range sensors. One on front and one on back. Knowing the I2C address conflict I changed the address of one with the core device discovery and assigned the new I2C address using the method provided with the range sensor class. When I go to read the distance it quires the address 0x28 instead of the address I set it to. Due to this there is an address conflict and I only get distance from the unchanged range sensor,

AlecHub, 2017-02-14:
For multiple I2C sensors, you have to use the `I2cDevice` interface class so that you can specify the I2C address of the sensor.  You also have to select `I2C Device` in configure robot.

```java
package org.firstinspires.ftc.teamcode;

/*
Modern Robotics Range Sensors Example
Created 10/31/2016 by Colton Mehlhoff of Modern Robotics using FTC SDK 2.35
Reuse permitted with credit where credit is due

Configuration:
I2CDevice "range28" (MRI Range Sensor with default I2C address 0x28
I2CDevice "range2a" (MRI Range Sensor with I2C address 0x2a

To change range sensor I2C Addresses, go to http://modernroboticsedu.com/mod/lesson/view.php?id=96
Support is available by emailing support@modernroboticsinc.com.
*/

import com.qualcomm.robotcore.eventloop.opmode.OpMode;
import com.qualcomm.robotcore.eventloop.opmode.TeleOp;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.I2cAddr;
import com.qualcomm.robotcore.hardware.I2cDevice;
import com.qualcomm.robotcore.hardware.I2cDeviceSynch;
import com.qualcomm.robotcore.hardware.I2cDeviceSynchImpl;

@TeleOp(name = "Range Sensors", group = "MRI")
//@Disabled
public class MRI_Range_Sensors extends OpMode {

    byte[] rangeAcache;
    byte[] rangeCcache;

    I2cDevice rangeA;
    I2cDevice rangeB;
    I2cDeviceSynch rangeAreader;
    I2cDeviceSynch rangeBreader;

    @Override
    public void init() {
        telemetry.addData("Status", "Initialized");

        rangeA = hardwareMap.i2cDevice.get("range28");
        rangeB = hardwareMap.i2cDevice.get("range2a");

        rangeAreader = new I2cDeviceSynchImpl(rangeA, I2cAddr.create8bit(0x28), false);
        rangeBreader = new I2cDeviceSynchImpl(rangeB, I2cAddr.create8bit(0x2a), false);

        rangeAreader.engage();
        rangeBreader.engage();
    }

    @Override
    public void loop() {
        rangeAcache = rangeAreader.read(0x04, 2);  //Read 2 bytes starting at 0x04
        rangeBcache = rangeBreader.read(0x04, 2);

        // Ultrasonic value is at index 0. & 0xFF creates a value between 0 and 255 instead of -127 to 128
        int LUS = rangeAcache[0] & 0xFF;
        int RUS = rangeBcache[0] & 0xFF;

        // Optical distance value is at index 1.
        int LODS = rangeAcache[1] & 0xFF;
        int RODS = rangeBcache[1] & 0xFF;

        // Display values
        telemetry.addData("1 L US", LUS);
        telemetry.addData("2 L ODS", LODS);
        telemetry.addData("3 R US", RUS);
        telemetry.addData("4 R ODS", RODS);
    }
}
```

an732001, 2017-04-15:
This does not work. The error is that it does not detect the sensors but the sensors are plugged in correctly. What is happening?

cmacfarl, 2017-12-04:
Believed to be fixed in https://github.com/ftctechnh/ftc_app/issues/360.  Feel free to reopen if not.

closed 2017-12-04 by cmacfarl