ftctechnh/ftc_app #327: Provide Driver Station app on F-Droid

qwertychouskie, 2017-02-14:
Google Play requires you to connect/make a Google account just to downloads apps.  F-Droid, on the other hand, is simple, small, does not need accounts, etc.  Having the DS app on F-Droid would be really useful.

NoahAndrews, 2017-02-15:
They would have to open source it first, so right now they can't add it. If they were to open source it and add it to F-Droid, I would be far, far more excited about the open sourcing part.

You know you can sideload the DS apk, right?

qwertychouskie, 2017-02-15:
Yeah, but then you have to manually update it...

NoahAndrews, 2017-02-15:
Sure, but that's still easier than updating the RC app, especially if there were any breaking changes in the API.

vwestin, 2017-02-16:
I have to say that it was simpler when they included the driver station APK in this project. It provided a clear way to be sure that the driver station matched the code the team was programming with. And it meant we did not have to do the work to download the APK to a laptop so that we could then load it to the phones. Maybe we can get that brought back....

qwertychouskie, 2017-02-16:
It is still in the project: https://github.com/ftctechnh/ftc_app/tree/master/doc/apk
It is just a pain to manually update it though...

EtherTyper, 2017-02-17:
Our team would greatly benefit from the DS app being open sourced.

vwestin, 2017-02-17:
Thank you. The team had overlooked it down there....

dvick, 2017-03-10:
It doesn't technically have to be open source if they were to make their own F-Droid repository. But, yeah, it would be nice if it were made open source.

guineawheek, 2017-03-10:
You could just cd to doc/apk and using install the app using adb:

`adb install [-r if reinstall] [-d if downgrade] FtcDriverStation-release.apk`

dvick, 2017-03-10:
That's exactly what I do to install the driver station. It would still be cool to see it on F-Droid.

JacobAMason, 2017-03-11:
Even easier/no adb required. Literally just plug in the phone, drag the apk into a directory (I usually just pick the Downloads folder), navigate to it on the device, and install it.