ftctechnh/ftc_app #329: `opModeIsActive` like functionality for `OpMode`?

EtherTyper, 2017-02-17:
It would be extremely useful to our team if you were to provide a method that signals whether the OpMode has terminated, since our OpModes rely on threading and interrupt models are proving unreliable. Just a suggestion, as we don't immediately require any such feature.

Thanks,
Eli Bradley

ethan-schaffer, 2017-02-17:
Do you need something aside from the provided 'opModeIsActive()' that is provided? You can also take a look at 'isStopRequested()'

Good Luck!

EtherTyper, 2017-02-17:
@ethan-schaffer Thank you!

EtherTyper, 2017-02-17:
@ethan-schaffer It appears `isStopRequested` is only available in `LinearOpMode` subclasses.

ethan-schaffer, 2017-02-17:
I can't think of an easy solution to that, sorry. OpModeIsActive should also work where you need it to, I believe. You can try putting the Linear OpMode into your Thread so you can call l.opModeIsActive(). If you want an example of code using these, I would be happy to write something up.

You could try having your teleOp as a linear opMode while using a while(opModeIsActive()) loop, I suppose.

calebsander, 2017-02-17:
The whole point of `OpMode` is to make the `loop()` method do a single update of the control system and then quit immediately. This allows for substantial optimizations like not having to check whether stop has been requested and not having to manually call `telemetry.update()`, as well as not having to ensure you are yielding enough time to the internal processes. If you are going to have a thread continually running in the background anyways, why not use `LinearOpMode`? It is also implemented by running the user program on a separate thread.

qwertychouskie, 2017-02-17:
Have you tried something like:
In the beginning, create a `var` `isStopping` and set it to false,
```
    @Override
    public void stop() {
        // Code here runs ONCE after the driver hits stop
        var isStopping = true;
    }
````
Then you can check whether isStopping is `true` or `false`

EtherTyper, 2017-02-17:
@calebsander Thank you for explaining.

@qwertychouskie Thanks, I will try that.

closed 2017-02-17 by EtherTyper