ftctechnh/ftc_app #330: Blocks-release.aar Break

Rydogdav, 2017-02-17:
Hey all. I was trying to run the gradle/master app at home on Android Studio to try and work on some code from home, however upon boot up and Build: Grade in the beginning, I consistently receive an error message, aborting the load. The error is "Execution failed for task ':FtcRobotController:prepareBlocksReleaseLibrary'.
> Could not expand ZIP 'C:\Users\rydog\Desktop\Robotics\ftc_app-master\libs\Blocks-release.aar'." Any known fixes?

olekmali, 2017-02-17:
Download the file from

1) 
https://github.com/MatthewMateusz/ftc9985-2016/blob/master/libs/Blocks-release.aar

or

2) https://github.com/ftctechnh/ftc_app/blob/master/libs/Blocks-release.aar

Good luck!

My students went through the same problem last November. The file 
somehow disappeared in the thin hard drive ;-)

Aleksander Malinowski
FTC Team #9985 Mentor
phone: +1 (309) 306 1131

cmacfarl, 2017-12-04:
Housekeeping

closed 2017-12-04 by cmacfarl