ftctechnh/ftc_app #332: HardwareMap name

leoforney, 2017-02-22:
It would be great if FIRST could provide a function for the HardwareMap that would return the currently loaded filename of the robot configuration in a String. By this, we can run one "TeleOp" code on our demo robot, and real robot without changing any code. It would just be a small but cool feature to see.

calebsander, 2017-02-22:
Why not simply write two different `OpMode`s and simply run the correct one for your robot?

ThePandaByte, 2017-02-23:
For my team's autonomous, we put it in a switch block, I think. We mapped a gamepad button, like dpad_up for example, which when pressed and selected during initialization, would make the robot do a specific set of tasks during the autonomous. We took a telemetry command to allow the robot to tell us which tasks it would do. So in a sense, it was like a bunch of opmodes inside one big opmode, if that makes any sense, and if that's what you mean.

leoforney, 2017-02-23:
You could write two different OpModes but the point is that if you can easily identify which robot the OpMode is running on. Also, this is an easily implementation, all it only has to read the file name and save it as a string in the HardwareMap object when it parses the xml file.