ftctechnh/ftc_app #339: I2C Bus Issues on DIM, OpMode Skips from Init to Stop, Adafruit IMU, I2cDeviceSynchImpl, Disengaging and Closing Sensors

benorgera, 2017-04-24:
Trying to get a consensus on the inexplicable I2c errors which my team (4102) and others have experienced when using many I2C sensors. Our I2C bus sometimes reaches a state in which registering any port-ready callback with the I2C controller (as is done when MR Range/Color Sensors, the BNO055 IMU or any other I2C devices are initialized/engaged) breaks the opMode out of the init and it jumps directly to stop without running. It seems that the first call engaging the range sensors (which we have been accessing through an I2cDeviceSynchImpl, so the I2C addresses can be specified) triggers the behavior, yet no errors are thrown so its hard to tell if it may instead be the IMU initialization.

Other teams have experienced this exact issue, as documented in this post (https://ftcforum.usfirst.org/forum/ftc-technology/7154-desperately-seeking-help-re-dim-i2c-sensor-issues), but no conclusive explanation or solution has been found. I have experimented with rewiring the I2C devices, replacing the DIM, updating the DIM to version 2.1, yet this problem intermittently persists. It seems restarting the phones and robot, and reinstalling the robot controller app may do the trick for a period, but at some point this state is once again achieved on the I2C bus, and no opModes can be run.

Other posts have hinted that maybe a faulty robot controller app build from Android Studio could be the culprit, but my hunch is that this has something to do with the I2C controller callbacks for each port. This issue only began a few days after we added code to disable the range sensors in autonomous once they are no longer needed, by calling `i2cDeviceSynchImpl.disengage()`, which deregisters the I2C port-ready callback for that device. It does not however deregister the end or begin callbacks for the I2C port, which is done with an `i2cDeviceSynchImpl.close()` call. Maybe by not explicitly closing, and deregistering all of the callbacks, on subsequent op mode runs the controller could callback a stale I2C device object from a previous opMode execution, and thereby cause this mysterious jump to stop behavior. 

According to this post from the dev team and its comments (https://ftcforum.usfirst.org/forum/ftc-technology/6806-gyro-read-latency-development-team-update?7724-Gyro-Read-Latency-Development-team-update=), engaging and disengaging I2C synchs is supported by the sdk, but no mention of closing devices between subsequent op mode runs is made. In this post about using multiple range sensors (https://ftcforum.usfirst.org/forum/first-tech-challenge-community-forum-this-is-an-open-forum/teams-helping-teams-programming/6594-problem-using-2-of-modernroboticsi2crangesensor-and-or-colorsensor) the I2cDeviceSynchImpl is initialized with isI2cDeviceOwned false, meaning the synch doesn't have the power to close the device its communicating with. Is this the right configuration to ensure I2C bus stability throughout multiple op mode runs? Should the synch have the ability to close the device it communicates with, or should the user even explicitly close the synch after the opMode ends/when they are done using the sensor?

(All of these questions apply to the I2C stability of the ModernRoboticsI2cRangeSensor class too, which implements an I2cDeviceSynch under the hood).

Any input whatsoever would be extremely helpful. Anyone with an understanding of the FTC I2C protocol please chime in on disengaging/closing sensors, and what really happens to an I2C device after an op mode stops and is run again. Many people are seeking clarity on how to most safely handle the DIM I2C devices, particularly between op mode runs, so that no erroneous state is produced on the I2C bus.

Thank you so much, your responses are greatly appreciated,

CHS Team 4102

cmacfarl, 2017-04-24:
Can you please provide a pointer to an opmode that causes this failure and robot logs from a failed run?

benorgera, 2017-05-13:
It seems these issues were most likely addressed in changes between v2.4 and v2.62. Although our robot controller was updated, our driver station was running an outdated version. Closing the issue, (but also glad to hear v3.1 has a new I2C implementation). Thanks

closed 2017-05-13 by benorgera

WH-2099, 2018-07-17:
We actually had the same problem with you.
We had fix the problem by updating REV Robotics Expansion Hub Firmware to 1.07.02.