ftctechnh/ftc_app #342: SDK Not Found

bhollers, 2017-05-12:
When updating to this newest version the gradle build fails with the following error:

Error:SDK location not found. Define location with sdk.dir in the local.properties file or with an ANDROID_HOME environment variable.

magneticflux-, 2017-05-12:
@bhollers That appears to be an issue with the way your Android Studio installation is set up. There are solutions on StackOverflow such as this one: http://stackoverflow.com/questions/27620262/sdk-location-not-found-define-location-with-sdk-dir-in-the-local-properties-fil or this one: http://stackoverflow.com/questions/19794200/gradle-android-and-the-android-home-sdk-location

cmacfarl, 2017-12-04:
Housekeeping

closed 2017-12-04 by cmacfarl