ftctechnh/ftc_app #345: Two sensors returning identical values with 3.1

slylockfox, 2017-06-06:
We have an opmode that reads two IR sensors.  One IR sensor has its address changed to 0x42.  The opmode worked fine, until we upgraded from 2.x to 3.1.  Now both sensors report exactly the same value, now matter where the beacon is.  It appears that this line is no longer working:
`irSeekerV.setI2cAddress(I2cAddr.create8bit(0x42));`

The opmode is here: https://github.com/slylockfox/ftc_app/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/UselessTestMRIrSeeker.java

JacobAMason, 2017-06-14:
This could be a duplicate of #325. It might be worth checking to see if the solution there works for you.

slylockfox, 2017-06-23:
It's not a duplicate of #325, as we have this working with release 2.4.  In fact I downloaded a fresh copy of 2.4 and added the opmode in, installed DS and RC on the same phones, and the opmode works as expected, showing different values from each sensor.

slylockfox, 2017-09-01:
@JacobAMason  Turns out my issue did overlap with #325 , apologies.  The fix for #325 fixed my issue too, as did the more thorough fix in #360