ftctechnh/ftc_app #347: I2C Configuration Impossibe on REV Expansion Hub

XykonV, 2017-06-10:
When I go to configure my robot's custom I2C sensor, I2C Device is not usable. All there is is I2C Device(Synchronous). Thus, I2C is impossible to use with the Expansion Hub.

XykonV, 2017-06-14:
Additional clarification: My code always fails to run because the configuration option for I2C devices in the REV module is I2C Device(Synchronous), and there is no option for just I2C Device. In the SDK, I2CDeviceSynch is just a reading interface. This doesn't really make much sense to me. Has anyone gotten I2C to work with the FTC SDK 3.1?
Code:
    I2cDevice pixy;
    I2cDeviceSynch pixyReader;
    I2cAddr pixyAddress = I2cAddr.create7bit(0x54);
    byte[] readCache;

    @Override
    public void runOpMode() throws InterruptedException {
        pixy = hardwareMap.i2cDevice.get("pixy");
        pixyReader = new I2cDeviceSynchImpl(pixy, pixyAddress, false);
        pixyReader.engage();

        waitForStart();

        while(opModeIsActive()) {
            readCache = pixyReader.read(0x54, 100);

            telemetry.addData("1", readCache[1]);
            telemetry.addData("2", readCache[2]);
            telemetry.addData("6", readCache[6]);
            telemetry.addData("7", readCache[7]);
            telemetry.addData("8", readCache[8]);
            telemetry.addData("9", readCache[9]);
            telemetry.addData("isInReadMode", pixy.isI2cPortInReadMode());
            telemetry.update();
        }
    }

rgatkinson, 2017-07-21:
The underlying semantics of I2cDevice are highly intertwined with the details of the I2c implementation on the Modern Robotics Core Device Interface Module, in particular the modality of reading vs writing and the timing of same with respect to a particular sequence of callbacks. Thus providing a fully correct implementation of I2cDevice on the Expansion Hub would have been a very significantly challenging undertaking. 

Further, in most cases, I2cDeviceSynch (which does both reading and writing) turns out to be an easier model to work with when accessing I2c devices. If there's functionality that's missing therefrom, please let us know.

XykonV, 2017-07-26:
Good to know it's intentional. Is there any sample code that you know of that uses I2cDeviceSynch? As you can see in my code, I use it, but I'm not sure how to initialize it.

rgatkinson, 2017-07-26:
Have a look at the subclasses of I2cDeviceSynchDevice (in Android Studio: 'Navigate' / 'Class' -> I2cDeviceSynchDevice, then click the little blue down arrow on the left beside the class name) such as ModernRoboticsI2cCompassSensor.

JorgeNava, 2017-11-16:
Hello, can somebody help me out with the programming of MR sensors on my Expansion Hub?

cmacfarl, 2017-11-17:
@JorgeNava General questions asking for help will get more traction if asked in either the FIRST forums or the FTC subreddit.  When asking please be as complete as possible describing what you want to do, what problems you are encountering, what platform you are using to develop software, etc...

closed 2017-11-17 by cmacfarl

JorgeNava, 2017-11-17:
Thanks @cmacfarl for the advise, I'll follow your suggestions for the future.