ftctechnh/ftc_app #348: Ftc Robot Controller can't find REV expansion hub

samduong01, 2017-06-13:
So I dowloaded the ftc robot controller via android studio yet when I click configure robot and scan for new devices, nothing shows up. Can someone help please.

XykonV, 2017-06-13:
There could be hundreds of things, we need more information. Try changing USB cables, ensure the module is getting power.

samduong01, 2017-06-13:
I did and yes the module is getting power the light is green

slylockfox, 2017-06-14:
Do you see Expansion Hub Portal 1?

slylockfox, 2017-06-14:
[image]

cmacfarl, 2017-06-14:
Are you running the 3.1 SDK?

samduong01, 2017-06-14:
No I do not see the expansion hub portal 1

slylockfox, 2017-06-14:
In my case, the light on the hub flashes blue-blue-green, which I think means the usb is connected and 12v power is connected.

cmacfarl, 2017-06-14:
It's the SDK that tells the REV module to flash that pattern, so it also means that the SDK is communicating with the controller.

Can you please post a screenshot of your robot controller version number?

samduong01, 2017-06-14:
[image: Inline image 1]

qwertychouskie, 2017-06-14:
@samduong01 Images have to be uploaded in the GitHub interface.

samduong01, 2017-06-15:
[image]

brandon0x52, 2017-07-08:
Over on issue #350 we solved this issue by [disabling Instant Run in Android Studio](https://ftcforum.usfirst.org/forum/ftc-technology/49831-sdk-3-1-robot-controller-app-cannot-find-rev-expansion-hub) by going to `File->Settings->Build, Execution, Deployment->Instant Run` and unchecking it. @samduong01, you should look at #350 and read through it to see if turning off this feature and building a fresh copy of the app helps!