ftctechnh/ftc_app #350: No USB Devices found on Robot Controller Scan

Keith503, 2017-06-22:
Installed latest 3.1 version of ftc_app on never before used Windows 10 laptop. No changes to Pushbot app, copied to team area and renamed to FF503_Pushbot_tank. Compiled clean and deployed to robot controller normally. Installed FTC driver station 3.1 from Google play store. Connected WiFi direct, phones are talking properly. When running USB scan on robot controller nothing returns. Puts up scanning box and then clears it. No USB devices are found or displayed. Using Modern Robotics controllers on working robot from last year. I have attached robot controller log file that looks like a null pointer exception after USB scan has completed and found 3 devices then tries to update UI. There are no configuration files on the robot controller, this is a first time install. Phones are a Motorola Moto G4. 
[FTC_Robot_Control_Log.txt](https://github.com/ftctechnh/ftc_app/files/1093288/FTC_Robot_Control_Log.txt)

richardelmore, 2017-06-24:
Seeing what sounds like the same issue.  Scan will display servo or device interface modules as long as no motor controllers are connected but anytime a motor controller is connected the scan shows no devices.  Switching back to the 2.5 app resolves the issue.

Armin-Mag, 2017-07-05:
Encountered the same issue.  The phone also cannot detect the REV module.

brandon0x52, 2017-07-07:
I have also purchased a new pair of Moto G4's and REV Robotics Expansion Hub. Although I cannot find the exact parameters necessary to replicate, sometimes my phone will not find the expansion hub at all, sometimes it will but I cannot select any motors, or sometimes my phone will throw the same error that appeared in Keith503's Robot Log. (`NullPointerException `on the `isDeviceFlavor` method)

magneticflux-, 2017-07-07:
@brandon0x52 Quick tip for pasting logs, you can copy the log and paste it into GitHub with nice formatting by surrounding it with "\`\`\`".

For example, to get this:
```
log log log
    test test
    test test
    test test
```
Type:
```
\`\`\`
log log log
    test test
    test test
    test test
\`\`\`
```
(Excluding the slashes)

brandon0x52, 2017-07-07:
Thank you @magneticflux-!

brandon0x52, 2017-07-07:
Although I cannot check this right now, I've searched the forum and found [this](https://ftcforum.usfirst.org/forum/ftc-technology/49831-sdk-3-1-robot-controller-app-cannot-find-rev-expansion-hub) thread. Instant Run has caused problems for my team in the past and I wouldn't doubt it being related to this issue. Is someone able to confirm this solves the issue?

magneticflux-, 2017-07-07:
@brandon0x52 I completely agree; while I love Instant Run for traditional apps and UIs, I'm certain that dynamically loading new dex files would cause some issues in the delicate setup FTC has going here. Last year I had my entire team disable Instant Run since we were seeing intermittent/non-deterministic issues with networking as well.

RedfishRobotics, 2017-07-07:
My team responded at the bottom of the linked thread, and yes, for that
specific issue it solved our problem immediately and we haven't had another
problem with the REV modules since.

Michael P Clark
Founder/Lead Mentor

Redfish Robotics, LLC
"We're Hooked on *FIRST*­®"

2017 FRC Bayou Regional Volunteer of the Year

http://www.redfishrobotics.com
Phone: 985-80F-TC10
Twitter: @FIRSTMentorMike <https://twitter.com/firstmentormike>

brandon0x52, 2017-07-08:
I've disabled Instant Run, did a clean build of the app onto my Moto G4 and ta-da! It finds the REV Robotics Expansion Hub. Since you started this issue @Keith503, try [turning off Instant Run](https://ftcforum.usfirst.org/forum/ftc-technology/49831-sdk-3-1-robot-controller-app-cannot-find-rev-expansion-hub) on Android Studio under `File->Settings->Build, Execution, Deployment->Instant Run` and let us know if it works.

Keith503, 2017-07-08:
Yes it worked. Disabled the instant run then a clean build and now it is able to scan and find all the motor controllers correctly!  Thank you to everyone who commented or helped identify and resolve the issue.

closed 2017-07-08 by Keith503

nywillb, 2017-09-14:
If anyone was wondering, the `Build, Execution, Deployment` menu is under `Android Studio>Preferences` on macOS. (I couldn't find it, so I thought others might have the same issue :smile:)

maylee0515, 2018-01-02:
Our team had the same problem with nothing showing up. However, turning off Instant Run doesn't seem to solve the problem. Any other suggestions?

NoahAndrews, 2018-01-02:
@maylee0515 can you post your log file?

maylee0515, 2018-01-03:
I don't have the log file on my computer but we had the same problem around September but it randomly started working and now its not working again.

maylee0515, 2018-01-03:
@NoahAndrews 
This is the log from the day it stopped working. Thanks for the help.
01-02 16:31:41.946 12848 12848 I chatty  : uid=10123(ps) expire 1 line
01-02 16:31:42.009 12898 12898 I chatty  : uid=10123(ps) expire 2 lines
01-02 16:31:42.011 12848 12848 I chatty  : uid=10123(ps) expire 1 line
01-02 16:31:42.078 12900 12900 I chatty  : uid=10123(ps) expire 2 lines
01-02 16:31:42.080 12848 12848 I chatty  : uid=10123(ps) expire 1 line
01-02 16:31:42.146 12903 12903 I chatty  : uid=10123(ps) expire 2 lines
01-02 16:31:42.147 12848 12848 W art     : Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a53 --instruction-set-features=default --dex-file=/data/app/com.qualcomm.ftcrobotcontroller-2/split_lib_slice_7_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_7_apk.apk@classes.dex) because non-0 exit status
01-02 16:31:42.208 12905 12905 E dex2oat : Failed to create oat file: /data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_8_apk.apk@classes.dex: Permission denied
01-02 16:31:42.208 12905 12905 I dex2oat : dex2oat took 522.969us (threads: 4) 
01-02 16:31:42.210 12848 12848 W art     : Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a53 --instruction-set-features=default --dex-file=/data/app/com.qualcomm.ftcrobotcontroller-2/split_lib_slice_8_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_8_apk.apk@classes.dex) because non-0 exit status
01-02 16:31:42.271 12909 12909 E dex2oat : Failed to create oat file: /data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_9_apk.apk@classes.dex: Permission denied
01-02 16:31:42.271 12909 12909 I dex2oat : dex2oat took 698.386us (threads: 4) 
01-02 16:31:42.273 12848 12848 W art     : Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a53 --instruction-set-features=default --dex-file=/data/app/com.qualcomm.ftcrobotcontroller-2/split_lib_slice_9_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_9_apk.apk@classes.dex) because non-0 exit status
01-02 16:31:42.296 12848 12848 I InstantRun: starting instant run server: is main process
01-02 16:31:42.319 12848 12848 V AppUtil : initializing: getExternalStorageDirectory()=/storage/emulated/0
01-02 16:31:42.321 12848 12913 V RobotCore: saving logcat to /storage/emulated/0/robotControllerLog.txt
01-02 16:31:42.321 12848 12913 V RobotCore: logging command line: logcat -f /storage/emulated/0/robotControllerLog.txt -r4096 -n4 -v threadtime UsbRequestJNI:S UsbRequest:S *:V
01-02 16:31:42.333 12848 12848 V ServiceStarter: attempting to start service MediaTransferProtocolMonitorService
01-02 16:31:42.335 12848 12848 V ServiceStarter: started service MediaTransferProtocolMonitorService
01-02 16:31:42.336 12848 12848 V ServiceStarter: attempting to start service OnBotJavaService
01-02 16:31:42.337 12848 12848 V ServiceStarter: started service OnBotJavaService
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3615): avc: denied { getattr } for uid=10123 path="/proc/1" dev="proc" ino=47688 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:init:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3616): avc: denied { search } for uid=10123 name="1" dev="proc" ino=47688 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:init:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3617): avc: denied { search } for uid=10123 name="1" dev="proc" ino=47688 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:init:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3618): avc: denied { getattr } for uid=10123 path="/proc/2" dev="proc" ino=47689 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:kernel:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3619): avc: denied { search } for uid=10123 name="2" dev="proc" ino=47689 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:kernel:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3620): avc: denied { search } for uid=10123 name="2" dev="proc" ino=47689 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:kernel:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3621): avc: denied { getattr } for uid=10123 path="/proc/3" dev="proc" ino=47690 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:kernel:s0 tclass=dir permissive=0
01-02 16:31:42.359 12915 12915 W ps      : type=1400 audit(0.0:3622): avc: denied { search } for uid=10123 name="3" dev="proc" ino=47690 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:kernel:s0 tclass=dir permissive=0
[... cut ...]
01-02 16:36:06.143 13368 13368 W art     : Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a53 --instruction-set-features=default --dex-file=/data/app/com.qualcomm.ftcrobotcontroller-2/split_lib_slice_3_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.qualcomm.ftcrobotcontroller-2@split_lib_slice_3_apk.apk@classes.dex) because non-0 exit status
01-02 16:36:06.193 13396 13396 E dex2oat : Failed to create oat file: /data/dalvik-cache/arm/

maylee0515, 2018-01-03:
@NoahAndrews 
This is the log from the day it stopped working. We also have the problem with only one light on our core power distribution module. We can't figure out if it is the module that is the problem or something else. 
[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/1600468/robotControllerLog.txt)

gearsincorg, 2018-01-03:
Does this line in the log file indicate that Instant run IS turned on?

01-02 16:31:42.296 12848 12848 I InstantRun: starting instant run server: is main process

gearsincorg, 2018-01-03:
Also, if you only have ONE light on your CPDM then you definately have a hardware problem.

There are 4 LEDs on that module, but one of them depends on the Phone being connected and active, but the other three are just power indicators.

Only 1 LED means that you are missing multiple voltages.    Time for a factory repair/replace.

NoahAndrews, 2018-01-03:
@gearsincorg Yeah, that's definitely a log with Instant Run enabled. I spent a lot of time looking at those recently. I actually have a fix that I need to show you guys.

maylee0515, 2018-01-03:
@gearsincorg Thank you for the help! We ordered a new one today. Hopefully it works.

DSmithChouteau, 2019-12-05:
Does this apply to onbot java also?   We're having the same problem and baffled.!

gearsincorg, 2019-12-05:
This thread is a year old, and quite convoluted..

What specific problem are you referring to?

magneticflux-, 2019-12-05:
Wow, what a necro!

I don't know @DSmithChouteau, why not try it and see how it turns out? It's definitely possibly for Instant Run to still cause issues, but it's gotten better over the past 2 years.

gearsincorg, 2019-12-05:
Instant Run is an Android Studio issue.  

It's extremely unlikely that it will impact someone using OnBotJava ...  since ....  they are not using Android Studio.

magneticflux-, 2019-12-05:
> It's extremely unlikely that it will impact someone using OnBotJava ... since .... they are not using Android Studio.

True! But if they were to deploy from Android Studio and then use OnBotJava, it *may* still cause issues because classpath scanning things might fail because the classes are split up for hot-swapping IIRC.

DSmithChouteau, 2019-12-05:
The configure robot scan is not finding anything and the Rev hubs are not showing up.   I have not idea why.. I am my team are new to Java and this problem.

gearsincorg, 2019-12-05:
@DSmithChouteau   For general "getting running" help, you should probably use the general FTC Tech forum. rather than this SDK Repo...    
https://ftcforum.firstinspires.org/forum/ftc-technology

To help us help you, we need to know more about your setup and what you have already tried...
There are some suggestions at that forum about what information to provide...

To get problem hardware connected I recommend the following sequence:

1) Disconnect your phone from the Expansion hub
2) Close and Swipe off the RC app, on your robot controller phone.
3) Power ON the expansion hub.
4) Plug the RC Phone into the expansion hub.

At this point, the App should auto launch (without you doing anything).  
If it does not, then you don't have a good connection between the phone and hub.

I would suspect your cables right away.  Are you using a good OTG cable, or could somone have swapped in a charging only cable.

Once you get the phone to auto-launch on connection, then y=the scan should work.

If you have two hubs, have you changed the address of one of them?  They must be different.

DSmithChouteau, 2019-12-05:
I ran across this when trying to configure the hubs, with no luck.  I guess
we need to download the latest firmware etc for the hubs...
http://www.revrobotics.com/software/

and then rename and configure the hubs separately..
https://github.com/ftctechnh/ftc_app/wiki/Using-Two-Expansion-Hubs

I have not heard of this before, but we will try it today.

D Smith