ftctechnh/ftc_app #353: Sdk is too low

gunrh1231, 2017-07-10:
Error:The SDK Build Tools revision (23.0.3) is too low for project ':FtcRobotController'. Minimum required is 25.0.0
<a href="openFile:E:/AndroidStudioProjects/ftc_app-master/FtcRobotController/build.gradle">Open File</a>

magneticflux-, 2017-07-10:
Replace "23.0.3" with "25.0.2" in some `.gradle` file.

gunrh1231, 2017-07-14:
Error:(23, 0) Could not get unknown property 'Focus' for repository container.
<a href="openFile:E:\AndroidStudioProjects\ftc_app-master\FtcRobotController\build.gradle">Open File</a>

gunrh1231, 2017-07-14:
now i got this error

magneticflux-, 2017-07-14:
@gunrh1231 Can you paste the `.gradle` file that is causing the error here using a code block? Just surround it with "\`\`\`".
Example:
\`\`\`
Code here...
\`\`\`
Becomes
```
Code here...
```

gunrh1231, 2017-07-15:
> I FIX IT THANK YOU

magneticflux-, 2017-07-15:
@gunrh1231 If you're sure it's fixed, you can mark the issue as solved using the "Close" button.

closed 2017-07-15 by gunrh1231

brandon0x52, 2017-07-18:
Just for clarification / help of anyone else who encounters this issue, can you explain how you were able to fix it, @gunrh1231?