ftctechnh/ftc_app #355: Robot Controller 3.1 not finding any Modern Robotics Devices

positronemitter, 2017-07-18:
When RC and DS apos updated to 3.1, they no linger would find and thereby not setup any of the Modern Robotics controllers. Downgrade to Ver 2.4 fixes issue.

brandon0x52, 2017-07-18:
Have you looked at the closed issue [#350](https://github.com/ftctechnh/ftc_app/issues/350)? Your issue and symptoms sound similar, I would try that and let us know what happens!

positronemitter, 2017-07-18:
Nearly identical problem but RC app Ver 3.1 works fine with Rev Robotics controller but not Modern Robotics. The instant run fix didn't seem to alter.

positronemitter, 2017-07-18:
Will retry the 3.1 again and see if fresh start now might work.

positronemitter, 2017-07-18:
Also, works fine with Instant Run deployed using ver 2.4.

brandon0x52, 2017-07-18:
Darn, that fixes most sporadic / uncommon issues such as this. Are you able to supply us with a copy the robot log from your RC with v3.1 on it? This way we can analyze and see if something is crashing the scanning process.
If possible, please provide a fresh one by deleting one if it exists and then running 3.1, scanning for hardware and then uploading that .txt log?

positronemitter, 2017-07-18:
Brandon,
   Deleted all and reloaded again with latest update 3.1. This time we ran
it with Instant Run and Extended Logs and submitted the logs to Google as
they request. Then we again uninstalled and reloaded with 3.1 but again
turned of Instant Run. This time its working like a champ. Can't thank you
enough. We probably would not have committed the time to trying it over if
it wasn't for your advice. Stay Awesome!

brandon0x52, 2017-08-05:
I'm glad I could help, @positronemitter! Also don't forget to mark the issue as closed if the problem is no longer persisting.