ftctechnh/ftc_app #360: Problem with I2cDeviceSynchImpl controller state machine

spencen, 2017-08-08:
There have been reports from users that have tried to use multiple instances of the same Modern Robotics (or any other) I2C sensor with different I2C addresses where the sensor setI2cAddress() method has no effect, so all sensors end up reporting the same value because only the default I2C address is actually being read. (See issues 345 and 325)

On further investigation the problem lies in the I2C state controller state machine in the I2cDeviceSynchImpl class.

The problem manifests on I2C sensors which use a readWindow with the mode set to REPEAT. When the I2C deviceClient.engage() method is called the controller state machine creates the command to read the I2C registers and enters the READ mode where it remains, reading I2C registers.
When the setI2cAddress() method is called it disengages, changes the I2C address, and then reengages. This unhooks and then re-hooks the controller, but it does not actually reset the controller state machine, so the state machine just picks up again in READ mode and continues reading the old I2C registers without rebuilding the command that sets the I2C address.

I believe the solution lies in resetting the controller state machine whenever the controller is unhooked.
I have performed limited testing but this seems to do the trick:

Change lines 292-312 of I2cDeviceSynchImpl.java:
```
    void attachToController()
    // All the state that we maintain that is tied to the state of our controller
    {
        this.readCacheTimeWindow = this.i2cDevice.getI2cReadCacheTimeWindow();
        this.readCache           = this.i2cDevice.getI2cReadCache();
        this.readCacheLock       = this.i2cDevice.getI2cReadCacheLock();
        this.writeCache          = this.i2cDevice.getI2cWriteCache();
        this.writeCacheLock      = this.i2cDevice.getI2cWriteCacheLock();

        this.nanoTimeReadCacheValid = 0;
        this.readCacheStatus  = READ_CACHE_STATUS.IDLE;
        this.initWriteCacheStatus(WRITE_CACHE_STATUS.IDLE);
        this.controllerPortMode = CONTROLLER_PORT_MODE.UNKNOWN;

        this.readWindowActuallyRead     = null;
        this.readWindowSentToController = null;
        this.isReadWindowSentToControllerInitialized = false;

        // So the callback will do it's thing to refresh based on the now-current window
        this.hasReadWindowChanged = true;
    }
```
to
```
    private void resetController()
    {
        this.nanoTimeReadCacheValid = 0;
        this.readCacheStatus  = READ_CACHE_STATUS.IDLE;
        this.initWriteCacheStatus(WRITE_CACHE_STATUS.IDLE);
        this.controllerPortMode = CONTROLLER_PORT_MODE.UNKNOWN;

        this.readWindowActuallyRead     = null;
        this.readWindowSentToController = null;
        this.isReadWindowSentToControllerInitialized = false;

        // So the callback will do it's thing to refresh based on the now-current window
        this.hasReadWindowChanged = true;
    }

    void attachToController()
    // All the state that we maintain that is tied to the state of our controller
    {
        this.readCacheTimeWindow = this.i2cDevice.getI2cReadCacheTimeWindow();
        this.readCache           = this.i2cDevice.getI2cReadCache();
        this.readCacheLock       = this.i2cDevice.getI2cReadCacheLock();
        this.writeCache          = this.i2cDevice.getI2cWriteCache();
        this.writeCacheLock      = this.i2cDevice.getI2cWriteCacheLock();

        resetController();
    }
```

and then at line 477 add resetController() so that it appears as:
```
                    this.isHooked = false;
                    resetController();
                    this.enableReadsAndWrites();

                    log(Log.VERBOSE, "...unhooking2 complete");
```

I am not certain that I fully understand the operation of this state machine, so some derivative of this fix may be more appropriate.

rgatkinson, 2017-08-09:
Thank you very much for this thorough analysis.

spencen, 2017-08-09:
No problem.
Any thoughts when a fix will be in a public release on gihub ?
Just asking so I know how to plan the software project start next month.

Thanks for all the work you put into it.

Nick

slylockfox, 2017-08-10:
Thank you!  #345 was mine, and I can test a fix whenever it's committed to the beta branch

slylockfox, 2017-09-01:
I can confirm that Nick's changes solved my problem with two IR Seekers.