ftctechnh/ftc_app #361: Driver Station App virus message

jrasor, 2017-08-16:
Downloading v 3.1 of the Driver Station app apk provokes a virus warning. Anti-virus: Windows Defender. OS: Win 10.
Downloading the WiFi channel editor app apk provokes no errors.

magneticflux-, 2017-08-16:
@jrasor I tried scanning both using Windows Defender and Malwarebytes on Windows 10 and it returned no warnings. If it *wasn't* a false positive, perhaps the `.apk` was injected with a self-replicating virus from elsewhere on your machine? Additionally, `.apk` files are not executable on Windows machines since they are Dalvik bytecode.

jrasor, 2017-08-16:
Thanks, Mitchell!

                I’m gonna download on a Win 7 workstation, with a different anti-virus, and let you know.

                I could download on Linux, and <gulp> just install from there onto the phone.

John Rasor SDB
Lead Coach 1, Don Bosco Tech FTC Team 5197 "the GearHeads"
[facebook]<http://facebook.com/DBTIG>   [twitter] <http://twitter.com/DBTI_G>

qwertychouskie, 2017-08-16:
@jrasor Why the "gulp"?  Just `sudo apt install adb` and `sudo adb install "/path/to/file.apk"`...
Or to avoid the `sudo` on the `sudo adb`, you can make a udev rule (https://github.com/M0Rf30/android-udev-rules).

jrasor, 2017-08-16:
Dear Mitchell,

                Downloaded the driver station app from https://github.com/ftctechnh/ftc_app/releases using my network workstation. It’s Win 7 pro, running on a Windows network. The network virus protection software is Sophos. This download went without issue, and the driver station app runs a robot correctly.

John Rasor SDB
Lead Coach 1, Don Bosco Tech FTC Team 5197 "the GearHeads"
[facebook]<http://facebook.com/DBTIG>   [twitter] <http://twitter.com/DBTI_G>

magneticflux-, 2017-08-16:
@jrasor If you still need to use the Windows 10 computer, I believe there are ways to override Windows Defender if it isn't downloading (maybe only happens with ~~~Internet Explorer~~~ Edge?) or if it is being quarantined.

jrasor, 2017-08-17:
Hi, Mitchell!

From: Mitchell Skaggs [mailto:notifications@github.com]
Sent: Wednesday, August 16, 2017 3:49 PM
To: ftctechnh/ftc_app
Cc: John Rasor; Mention
Subject: Re: [ftctechnh/ftc_app] Driver Station App virus message (#361)

If you still need to use the Windows 10 computer, I believe there are ways to override Windows Defender if it isn't downloading (maybe only happens with Internet Explorer Edge?) or if it is being quarantined.

                OK, I’ll look into that. I’ve got the driver app 3.1 running after a successful download, so this second look is not high priority.

                QwertyChouskie had a Linux tip; I’ll also try that.

John Rasor SDB
Lead Coach 1, Don Bosco Tech FTC Team 5197 "the GearHeads"
[facebook]<http://facebook.com/DBTIG>   [twitter] <http://twitter.com/DBTI_G>

jrasor, 2017-08-18:
@qwertychouskie 
 
> Just sudo apt install adb and sudo adb install "/path/to/file.apk"...
  That worked. Of course, one needs the device attached. Thanks! Closing issue.

jrasor, 2017-08-18:
Thanks, all. Two solutions:
1.  Download from network workstation, Win 7, network protected by Sophos.
2.  Download and install from adb on a Linux workstation.

closed 2017-08-18 by jrasor