ftctechnh/ftc_app #364: Robot Controller cannot open REV Robotics USB Expansion Hub Portal

NickDeGrood, 2017-08-26:
While working on a minibot, the phones displayed this error, and it prevented me from running the program. The error says "Warning: unable to open REV Robotics USB Expansion Hub Portal [DQ15ZH7N]." To fix it, I tried restarting the phones/robot, changing the battery, and replacing the cable, but it still persisted. I noticed that it doesn't display the voltage sensor and the INIT button is grayed out, similar to when the phone isn't plugged in to the robot. Because of this, I'm guessing that the Robot Controller is having issues communicating with the REV Expansion Hub. Anyone know how to fix this? Attached is an image of the RC and the DS with the error.

cmacfarl, 2017-08-26:
This does indeed indicate that the robot controller can not communicate with the expansion hub.  Can you perhaps grab the robot controller log?

gearsincorg, 2017-08-26:
I know this feedback isn't an immediate help, but it is a suggested "lowest
common denominator" test.

1) Unplug the phone from the Hub, and turn off the robot and phone
(actually power the phone down).
2) Turn the RC phone back on again, but do not start the RC app.
3) Plug the phone into the unpowered Expansion hub.

Two things should happen...

A) The hub LED should light up, and flash a color pattern
and
B) the Phone should automatically launch the RC app.

Do any, either, or both of these two things happen?

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

NickDeGrood, 2017-08-26:
Some LEDs did turn on, but the phone did not launch the RC app.

qwertychouskie, 2017-08-26:
Make sure you have "Instant Run disabled in Android Studio, and re-download the RC app.

jasonrahm, 2017-10-17:
Instant run disabled in android studio and I still get this very consistently, but not always. Seems a rain dance and rebooting and reconnecting everything occasionally fixes this problem.

magneticflux-, 2017-10-17:
@jasonrahm Perhaps a faulty USB cable or connector? Are you on the latest version on both phones? The phones might also need to be restarted because the USB driver sometimes breaks. Are the phones themselves updated?

canIplay100, 2017-10-19:
I have done the steps for what gearsincorg did and it won't work, please help.

robolords, 2017-11-19:
Hi All, our team is also facing the same issue. Are there any updates? If you were able to fix it, could you please share the steps?

Team8983, 2017-12-05:
@gearsincorg Thanks for the suggestion.  Our team is struggling with an issue similar to the one posted here.  We have tried multiple different cables, different phones, power cycles, software updates, etc.  We are still unable to open Rev Robotics USB Expansion portal.  The led lights when the driver station is plugged in but not the robot controller.  Any help would be GREATLY appreciated.

gearsincorg, 2017-12-06:
@Team8983  

The error message shown is trying to access a specific REV hub.    This means that the robot it trying to connect to the hub that is listed in the current configuration.

Since many teams have more than one hub (spares etc), is there a possibility that you are trying to connect to a different hub?

Have you tried just going into the Robot Configuration screen and doing a "New" scan?  This will show ANY hubs that are present.  If it finds a different hub, you may just be able to go back to your current configuration and use the "FIX" button to switch USB addresses with the current hub.

Phil.

ablackmond, 2017-12-06:
We are having this connection problem, too, but working with the block coding, not android studio.  Any suggestions?
We got it to work one day last week but have not been able to get it to connect since.

premkk, 2017-12-09:
We were having the same problem. We tried the suggestion by @gearsincorg and worked great.

geras5070, 2018-11-21:
I have tried the solutions provided above. 
There is an amber lights when the phone is plugged in. When the power is turn on then the light is blue. The phones are connected but the unable to open the REV Robotics USB Expansion Hub portal appears. We have changed cables. Any additional suggestions.

cmacfarl, 2018-11-21:
@geras5070 Have you tried this suggestion from Phil?

> Have you tried just going into the Robot Configuration screen and doing a "New" scan? This will show ANY hubs that are present.

gearsincorg, 2018-11-22:
Are you running one or two hubs?

If you have two, what are their addresses?  If the parent hub (connected to the phone) has an address greater than the child (daisy chained) hub there may be a problem if your App revision is less than 4.3.

Best if you update your app version, and/or swap the order of the hubs.

NoahAndrews, 2018-11-22:
Note that you may need to re-configure the robot after updating in order to fix the problem @gearsincorg mentioned