ftctechnh/ftc_app #366: MotorConfigurationType is very difficult to mock

nitrocalcite, 2017-08-30:
Because `MotorConfigurationType` is a final class, it cannot be mocked without the use of bytecode manipulation like in PowerMock.  `MotorConfigurationType` has a public constructor, but how exactly it fills in fields like `ticksPerRev` `gearing` `maxRPM` etc is very unclear (to me anyway).  The primary method of obtaining instances seems to be through the static method `getMotorType`, but that poses its own problems from a mocking standpoint since it relies on `UserConfigurationTypeManager`.  Additionally, `MotorConfigurationType` does not implement a single interface, which means I can't create a mockable class that implements that interface and delegates its functions to `MotorConfigurationType`.

Has anyone else found a way to mock MotorConfigurationType?  I'm attempting to be able to `getTicksPerRev` (among other things) from a unit test, but can't figure out how to circumvent this issue.  

Alternatively, can a `IMotorConfigurationType` interface be created, so that creating a mockable class which implements that interface is possible?  I assume that it can't be declared non-final due to "serialization concerns"*

*from a comment in `MotorConfigurationType.java`

magneticflux-, 2017-08-30:
As far as I know, Mockito has been able to mock final classes for a while. [Here](https://stackoverflow.com/questions/14292863/how-to-mock-a-final-class-with-mockito/40018295#40018295) is a post explaining it. That was implemented in `2.1.z`, and Mockito is now on `2.8.z` iirc

nitrocalcite, 2017-08-31:
For some reason I thought 1.10.19 was the latest, and that Mockito 2 was just rolling out.  I'm very wrong.  
Thank you!  Apologies for missing that lol.

closed 2017-08-31 by nitrocalcite