ftctechnh/ftc_app #368: REV Expansion Hub unable to communicate with IMU on 3.1

nitrocalcite, 2017-08-31:
There seems to be an issue with reading from the IMU that may be related to `I2CDeviceSynch`.  When initializing the IMU, it returns 0-length data instead of 26-length data as it should be.  This is not an issue with I2C bus 0, as a REV color sensor connected to that bus works properly.  However, `LynxI2cColorRangeSensor` uses `I2CDeviceSynchSimple` instead of `I2CDeviceSynch`.  We have not yet tried an external IMU as we _might_ have fried ours already.  We have confirmed this behavior across 2 different expansion hubs: one from FTC Storefront and one straight from REV.  

The error is below.  The full log is [here](https://gist.github.com/nitrocalcite/3047f541701d88caaacc842cb0ee708b).
```
08-31 18:36:45.152 16279-16349/com.qualcomm.ftcrobotcontroller V/RobotCore: addr=false data=false arb=false clock=false
08-31 18:36:45.152 16279-16349/com.qualcomm.ftcrobotcontroller E/LynxI2cDeviceSynch: readStatusQuery: cbExpected=26 cbRead=0
```

cmacfarl, 2017-08-31:
Certainly looks strange.  Can you please push your opmode to someplace on github?

nitrocalcite, 2017-09-01:
Unfortunetly I'm not the owner of our currently-private repo, and we are relying on an intermediate library we wrote to access the SDK, so the opmode itself probably won't be of much use.  I'll talk to the owner to see if I can get that changed.

With that being said, there basically isn't much we are doing special with the IMU.  We are using `LinearOpMode`, and in `runOpMode`, we call `hardwareMap.get(BNO055IMU.class, "internalIMU")`.  We hand this off to an intermediate class, but as of right now, that class implements `BNO055IMU` and just delegates everything to the IMU object we get from `hardwareMap`.  Then, after the play button is pressed, we have `imu.getAngularVelocity().xRotationRate` printed to telemetry.  

When we get this error, we get a telemetry message saying "Warning: problem with 'internalIMU'" on both the RC and DS.  This message appears when we press play.  It only disappears if we restart the app.  This behavior is consistent.  

We do pretty much the exact same thing with the REV color sensor, with `colorSensor.blue()` printed to telemetry instead.

magneticflux-, 2017-09-01:
@nitrocalcite Maybe a minimal test case would help? If possible, starting from scratch (without any libraries) and still reproducing the bug as simply as possible would probably rule out any interference.

nitrocalcite, 2017-09-01:
@magneticflux That's a very good idea.  I'm working on one right now, and my teammate is working on copying the repository into a public one at the same time.

nitrocalcite, 2017-09-01:
Our repo is [here](https://github.com/branvu/dbzCode18).  The opmode we are running is `TestOpMode`.
The actual instantiation of the sensor is handled by `DbzHardwareMap` through `DbzOpMode`.

nitrocalcite, 2017-09-01:
Also, here is a short test opmode that reproduces the problem for me:
```
@Autonomous(name = "RevImu", group = "Tests")
public class RevImuOpMode extends LinearOpMode {
    @Override
    public void runOpMode() throws InterruptedException {
        BNO055IMU imu = hardwareMap.get(BNO055IMU.class, "internalIMU");
        ElapsedTime runtime = new ElapsedTime();

        waitForStart();
        runtime.reset();

        while (opModeIsActive()) {
            telemetry.addData("imu x rotation", imu.getAngularVelocity().xRotationRate);
            telemetry.addData("Status", "Run Time: " + runtime.toString());
            telemetry.update();
        }
    }
}
```

magneticflux-, 2017-09-01:
@nitrocalcite FYI, you also have the option to transform an existing repository to a public one, if you want to preserve more history (issues, PRs, ownership? idk) than through a copy-paste. It's under Repo Settings -> Danger Zone. I don't know your situation so I don't know if that's suitable, but it's an option.

That test looks pretty minimal, so I doubt the issue lies in any libraries you're using. Just to cover all possibilities, are all of your wires are plugged in to the correct ports and the correct direction?

nitrocalcite, 2017-09-01:
@magneticflux Good to know, thanks.
I only have the REV Color Sensor plugged in (in addition to power & USB to the phone).  Additionally, the IMU is internal to the module, so I doubt that's the issue.  It's still a problem if I remove the color sensor from bus 0.

nitrocalcite, 2017-09-17:
We eventually fixed this issue by assigning parameters for the IMU.  The test code doesn't do that.

However, I feel like the FTC SDK should have used default parameters (with a warning maybe?) or at least thrown an error to indicate this issue.  Going from those logs all the way to the solution was quite a jump.

cmcknight, 2017-09-17:
Could you share those values?

nitrocalcite, 2017-09-17:
@cmcknight We ran this:
```
        BNO055IMU.Parameters parameters = new BNO055IMU.Parameters();
        parameters.loggingEnabled = true;
        parameters.loggingTag = "Imu";

        imu.initialize(parameters);
```

cmcknight, 2017-09-17:
Thanks! The boys on the team are just starting to work with the IMU for the first time and I know they’ll appreciate the information and help!