ftctechnh/ftc_app #370: Changing connection mode between two phones?

Candymini, 2017-09-07:
During a game, we sometimes find wifi direct is pretty unstable. Sometimes the phones disconnect for no reasons, and other times phones other than our driver station may request connection to our robot controller. Is it possible to develop a more stable connection mode? Like bluetooth?

gearsincorg, 2017-09-07:
Trust me...   anyone who was around in the first years of FTC will tell you
that Bluetooth was NOT stable, and were so happy when we switched to WiFi.

The problem is that many technologies that work fine when there are 5 or 10
connections, fail miserably when there are 100 or 200.

An FTC competition (especially with 2 or more fields) is a very challenging
environment.

The way to develop a more stable environment is to get good diagnostic data
from the field.
As a competitor, you can help by doing several things:

1) Be consistent in how you use your system.  If you have problems, try to
notice what else is happening.  It's unusual for dropouts to happen for "no
reason".  There is usually a reason, we just may not know what it is YET.
The more we can know about it the better, so FIRST can improve stability.
Record your observations.  Teach your team members to be good observers.

2) If you notice a consistent failure sequence or condition... record it in
as much detail as possible so that the Tech Team can try to recreate it.
This is what I do, and it makes it easier for others to correlate and fix.

3) The phones store log files, that record a lot of STUFF.  If you are
having problems, try to make these log files available to the FTA/CSA at an
event.  The log files are in the phone's root folder, so if you connect
them to a PC and put the phone in media mode, you can just copy them off
via USB.  What I do is use the phone's File Manager (download an app if you
have a Motorola phone) to delete the log files at the start of each
practice/competition day so they start fresh.  It also helps if you connect
your phones to the internet temporarily to let them synch their times.

The Tech Team aren't omnipotent, and it takes a long time to get any system
to be stable with limited resources.  Be a resource...

Phil

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

emmell620, 2017-09-07:
Good advice Phil.  One other thing to check is to see if there is an active WiFi Detective program running near you that actively kills random WiFi connections. Schools are very good about installing these to prevent rogue WiFi networks popping up and interfering with the school's WiFi.

qwertychouskie, 2017-09-07:
Also:

1. What phones are you using?  The ZTE Speeds are awful...  The realistic minimum is one of the Motos. (https://www.amazon.com/Moto-G-Play-Verizon-Prepaid/dp/B01M61A8JK/)
2. Where in the robot is the phone mounted?  Having the phone mounted in the robot will block connections (as we learned the hard way at Worlds).
3. Are you using the latest version of the SDK? (3.4 as of this writing)

gearsincorg, 2017-09-07:
> Office: 301.387.2331 <(301)%20387-2331>
> Mobile: 301.501.7424 <(301)%20501-7424>

PS, if you see another phone trying to connect to your phone, it's a person
doing it, not WiFi direct.

It could be happening by accident, or on purpose, but it does not effect an
existing connection unless you accept.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

gearsincorg, 2017-09-07:
>    G-Play-Verizon-Prepaid/dp/B01M61A8JK/

I wouldn't say that the ZTE Speeds are awful. Just not great :)
I would agree that if you are doing vision processing, you definitely
notice the Moto G improvement, but otherwise, they are pretty robust.
The ZTE's have the advantage that you can do wireless code downloads more
easilly.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

qwertychouskie, 2017-09-07:
Uh...  Try using one at Super-Regionals or Worlds.  Even the Moto G4 Play had issues at Worlds until we moved it to the outside of the robot.  Trust me, the G4 Play is well worth the $42.

NoahAndrews, 2017-09-07:
> The ZTE's have the advantage that you can do wireless code downloads more
easilly.

What about the G4 makes that harder?

cmacfarl, 2017-09-07:
@qwertychouskie My team used ZTE Speeds at both Super and Worlds this year.  We had trouble pairing in the venue at Supers, but taking the phones outside solved that problem.  Once paired we had no connection problems.   We had no connectivity problems at Houston Worlds.

@NoahAndrews The ZTE's allow you to be paired with another phone _and_ connected to a local wifi access point at the same time.  The G4 does not support this and so you have to constantly swap back and forth if you want to use adb wireless to deploy software to the phone.

AlecHub, 2017-09-07:
There is a setting in the Wi-Fi Direct settings that allows you to **_"Limit number of Wi-Fi Direct devices to improve performance"_**.  

Configure the RC phone (the group owner) to limit the number of devices in the group to 2 devices (the DS and the RC).  After your DS connects to your RC, no other teams will be able to connect to your RC.  Limiting the Wi-Fi Direct group to 2 devices may also help improve Wi-Fi Direct performance as the setting suggests.

qwertychouskie, 2017-09-07:
@cmacfarl I'd still highly advise against it.

Candymini, 2017-09-08:
Thank you so much for your advise! Sometimes, especially during a regional final last year, our RC was connected so frequently that the performance of the robot was more or less affected . In fact, not only our RC, but all four in the game were interrupted. Our robot last year was not affected greatly, but some other robots disconnected at the beginning of the game and failed to reconnect until the end. The reason I opened the issue is just want to see if there is any solution. Your suggestions are amazing, we will try them this season!

Candymini, 2017-09-08:
@qwertychouskie We use neither ZTE or Motos. We use Mi Phone, which our sponsor provides. The connection performance is not consistent as far as I know. Last season we use sdk 2.4 for stability, but this season we will switch to 3.4.

closed 2017-09-08 by Candymini

qwertychouskie, 2017-09-08:
@Candymini I don't think that is an allowed device... From the Game Manual part 1:
```
Driver Station consists only of one Android device: 
ZTE Speed, Motorola 
Moto G 2nd
Generation, Motorola Moto G 3rd
Generation, Motorola Moto G4 Play, Google Nexus 5, or 
Samsung Galaxy S5.
<RE06> 
<RE16>a
Robot Controller
consists only of one Android device: 
ZTE Speed, Motorola Moto G 2nd
Generation, Motorola Moto G 3rd
Generation, Motorola Moto G4 Play, Google Nexus 5, or 
Samsung Galaxy S5.
The Android device’s USB interface only connects to the Core Power 
Distribution Module
or the REV Expansion Hub.
```

Candymini, 2017-09-08:
@qwertychouskie I think it may be a special rule for China regionals. Our sponsor is pretty authoritative and a lot of team uses it.

qwertychouskie, 2017-09-08:
OK, just keep in mind that if you advance, you may have to get new phones.

gorpong, 2017-09-08:
@qwertychouskie, @Candymini is correct in that this is a special rule for China-based teams.  They were allowed the Mi devices at last year's Houston Championship (I was one of the inspectors) because they are the only devices with snapdragon processors that are available to those teams.

qwertychouskie, 2017-09-08:
Interesting, learn something new every day.  Seems weird that it was not mentioned in the game manual though.

gearsincorg, 2017-09-08:
Probably was in the Chinese version....  (Just kidding)