ftctechnh/ftc_app #371: SetVelocity not working with Matrix 12V Motor on 3.1

branvu, 2017-09-07:
When we `setVelocity` of the motor to 0 or 1, the motor seems to be working. However, for values in between 0 and 1, the motor rapidly pulses between stop and forward. We think the cause of this is from the PID tuning in the SDK, but we are not sure. 

Has anybody else observed this before?

cmacfarl, 2017-09-07:
Which controller are you using and how did you configure the motor in the robot controller?

branvu, 2017-09-07:
When we zero'd out the P coefficient, the issue goes away.
We are using the REV controller and it is configured as a Matrix 12V motor, which corresponds with our physical Matrix 12V motor.

qwertychouskie, 2017-09-07:
Also try upgrading to SDK version 3.4.

gearsincorg, 2017-09-08:
Are you really using SetVelocity() ?  
An input of 0-1 sounds more like setPower()

Can you show how you are using it?

void setVelocity(double angularRate,   AngleUnit unit)
Sets the velocity of the motor
Parameters:
angularRate - the desired angular rate, in units per second
unit - the units in which angularRate is expressed

nitrocalcite, 2017-09-09:
We are indeed using `setVelocity()`.  However, I think @branvu made a mistake on the values.  We picked 0 m/s and 1 m/s as convenient values that should be well within the range of our robot.  We divide this by our wheel radius in meters (~0.05) and use that as the argument to `setVelocity`.  So, we are calling this method in the main loop (`while(opModeActive())`):
```
leftMotor.setVelocity(gamepad1.left_stick_x/0.05, AngleUnit.RADIANS);
```
So, the actual values we are passing to `setVelocity` are between 0 and 20 rad/s.  Apologies for the earlier mistake.

gearsincorg, 2017-09-09:
No problems.

Let us know how it works out once you tame those values.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

nitrocalcite, 2017-09-09:
@gearsincorg Is there an issue with passing values from 0-20 rad/s into `setVelocity`?  I understand 20 rad/s is like, 3 rev/s, and might be too fast for the motor, but the issues we are encountering are in the neighborhood of 4-10 rad/s.  The specific values that the jerking motion starts at is very inconsistent, depending on how we move the joystick (ie slamming it quickly to one side vs slowly ramping it up), but that 4-10 rad/s reflects the region that we most commonly have issues.

nitrocalcite, 2017-09-10:
@qwertychouskie Updated to 3.4, did not fix the issue.

gearsincorg, 2017-09-10:
I know it's not a solution, but have you tried a different motor type in
your configuration file?
It's possible that the matrix speed/PID values are not correct.  This would
at least point to a culpret.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

nitrocalcite, 2017-09-10:
As @branvu said, we were able to fix the issue by setting the P coefficient for the PID loop to 0.  The efficacy of this is reproducible.  Just for kicks, we tried zeroing out the I and D coefficients, but the problem remained.  We've not yet tried _reducing_ the P coefficient.

gearsincorg, 2017-09-11:
The real question is: with a PID of zero, was the SetVelocity() actually
working?

That is:  did it control the velocity according to the requested value, and
did it stay constant with a varying load.

I think to test this properly, you would need to disconnect the motors form
the joystick, and feed in a constant SetVelocity() value based on a button
press.  eg: 3.141 radians per second,
and then see of the speed held constant as you put some friction on the
wheel.

Without actuclly testing this, you won't know for sure that it's actually
working properly.
Without doing this test, you may as well just use SetPower().

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

nitrocalcite, 2017-09-11:
@gearsincorg I see what you are saying, and I can test this at some point.  I was thinking of cross-checking the motor's encoder by connecting another unpowered motor and monitoring its encoder.  I could also load the powered motor by shorting (or connecting a resistor) the terminals of the unpowered motor.

rgatkinson, 2017-09-11:
Hmmm... which REV motor port are you connected to? If you're connected to ports 0 or 3, does the symptom change if you instead connect to port 1 or 2?

nitrocalcite, 2017-09-11:
@rgatkinson I can't remember with 100% accuracy, but I'm pretty sure that we observed this behavior on two motors: one connected to 0 and one connected to 1.