ftctechnh/ftc_app #373: null references in java :shocked:

fwprobotics, 2017-09-09:
error is thrown on app startup, robot status is `EMERGENCY STOP`
restarting robot does not work

```
Error: unable to create robot: Robot failed to start: Attmept to invoke interface method 'boolean com.qualcomm.robotcode.hardware.configuration.ConfigurationType.isDeviceFlavor(com.qualcomm.robotcode.hardware.configuration.UserConfigurationType$Flavor)' on a null object reference
```

qwertychouskie, 2017-09-09:
This sounds like an issue with your code, not the SDK itself.  Try posting on the forum.

knxtdoor, 2017-09-09:
We had this issue, disabling instant run in Android Studio fixed it for us.

devsnek, 2017-09-09:
@qwertychouskie i don't call those methods anywhere in my code or i would have checked that first

@knxtdoor thanks i'll try that when i'm in the shop next (on tuesday)

brandon0x52, 2017-09-10:
I agree with @knxtdoor, if you take a look at #350 that is the same null reference I was getting. Turning off Instant Run in Android Studio was the fix for me as well. Let us know how it goes!

devsnek, 2017-09-10:
awesome i'll close this (or not 👀 ) after i test

cmacfarl, 2017-09-10:
Instant run is very problematic.  The official line is that all teams using AS should turn it off.  Unfortunately with Google making the decision to enable it by default, this is hard to accomplish.

fwprobotics, 2017-09-11:
it worked, thanks

closed 2017-09-11 by fwprobotics